OpenDroneMap/ODM #384: Set of images produces "collage" like distorted orthophoto

pierotofy, 2016-09-26:
Input images: https://masseranolabs.com/images.zip (taken at 50 meters AGL, 80% overlap). Trees in the area likely lead to poor features detection.
Camera: DJI Phantom 3 (ccd: 6.16)

Processed using dev branch, default options.

Resulting orthophoto: 

[image]

Changing option "--matcher-distance" to 50 yields:

[image]

GPS EXIF data looks correct:

[image]

@paulinus

dakotabenjamin, 2016-09-26:
This looks like an issue with OpenSfM. Look at the sparse cloud here: [image]

dakotabenjamin, 2016-09-29:
For the sake of transparency, it was suggested on the [gitter](https://gitter.im/OpenDroneMap/OpenDroneMap) that this may be a camera calibration issue. 

I've produced a kludge-y calibration tool [here](https://github.com/dakotabenjamin/CameraCalibration) that produces calibrated photos based on distortion parameters from checkerboard reference photos.

pierotofy, 2016-09-29:
I'll try to run the tool and retry processing. I'll report results here. Thanks @dakotabenjamin

pierotofy, 2016-09-29:
I ran the CameraDistortion tool using the calibration images for DJI Phantom 3, and I got this:

[image]

Point cloud (shows part of one of the houses, which makes sense, since there are more features there).

[image]

dakotabenjamin, 2016-09-29:
That's disappointing. It looks like there are a few problems going on here
1. You have a partial reconstruction. This _may_ be solved by increasing the number of features extracted or, oddly enough, decreasing the resize value. 
2. Since the point colors in the point cloud are fine I'm wondering if it's a texturing bug. 

Quick question, what time of day were these photos taken? It almost looks like there are lots of shadows in the point colors of the point cloud.

pierotofy, 2016-09-29:
On 1. original images were 4000x3000 but I set a resize parameter of 1000.

Images were taken around 5:25pm, so just before twilight. Not ideal light conditions I realize. I would expect some distortion, but the output is really messed up.

dakotabenjamin, 2016-09-29:
I'm at a loss at this point. I'm hoping @paulinus can provide any insight, although if its a texturing problem, @smathermather we should ping spotscale.

pierotofy, 2016-09-29:
Result of processing with DroneDeploy: http://drdp.ly/2XlFAs

Some distortion near trees, but main features are reconstructed.

smathermather, 2016-09-30:
Resize parameter of 1000 is pretty low, and if you didn't change your matching parameters, this could result in a smaller point cloud. @pierotofy: have you run this with undistorted images and with defaults?

pierotofy, 2016-09-30:
@smathermather just tried with all defaults parameters and undistorted images.

[image]

smathermather, 2016-10-09:
@pierotofy -- what does your textured mesh look like?

pierotofy, 2016-10-09:
Here it is (default parameters):

[image]

Doesn't look very good.

smathermather, 2016-10-09:
Hard to tell for sure-- does the textured mesh look fine but the orthophoto is problematic?

pierotofy, 2016-10-09:
Well, the orthophoto is definitely problematic, the textured mesh is not complete, it displays only a very small area (like, 4 tiles instead of  the 200+ from the dataset) and elevation is also not properly represented (look at the warp in the image above).

Images are linked at the top of this discussion for anyone that wants to reproduce this problem.

dakotabenjamin, 2016-10-11:
Hey @xialang2012 I ran your images. Can you tell me what parameters you used?

dakotabenjamin, 2016-10-20:
@pierotofy have you tried a resize of 1200 or 1500? It looks like on #392 it's being linked back to a too-small resizing.

pierotofy, 2016-10-20:
I've tried using 2400 (the default), but I still get malformed output.

I wish I had more time to dig into the code myself to troubleshoot it, but I'm busy with WebODM.

xialang2012, 2016-10-21:
@pierotofy ,I have test the dataset that you provide with the default parameter, and the result is reasonable

[image]

pierotofy, 2016-10-21:
O_o

Is it be possible that a computer with low RAM is the cause? I ran my dataset with a VM that had only 3 GB of RAM.

xialang2012, 2016-10-21:
I think you should update to the lastly version of the dev branch and try it again.
I think it is none business of the RAM size.

For my experience the weird ortho image would be product when number of the input images is small, such as less than 8 and with the resize to 1000(about). Besides, I found only when we used the mvs-texting to texture the mesh, it will occurred.

pierotofy, 2016-10-22:
Ok I can confirm that using the latest code in the dev branch seems to work. I got this results (resized the images to 1800, default parameters).

[image]

[image]

This can probably be closed! Thanks everyone for the help.

closed 2016-10-22 by pierotofy

JLMarcelino, 2024-05-22:
Hi! Sorry for bothering here, but I have the exact same problem (2024!). Can  you explain how can I update to the lastly version of the dev branch?

smathermather, 2024-05-22:
This was a walk through memory lane to see this issue. Hi @pierotofy -- did you ever figure out how to improve your results from OpenDroneMap? :smile: 

@JLMarcelino -- please come over to community.opendronemap.org where we provide support and troubleshooting. We try to keep the repositories focused on bug reports and improvements, but we'll get you sorted over on the community forum.