OpenDroneMap/ODM #385: ODM path defining for starting python run.py in terminal to process images

pahpad, 2016-09-28:
hi. I am beginer in ubuntu and ODM. I installed ODM in OpenDroneMap in home directory. its path is "  /home/administrator/OpenDroneMap "  and there are some files like run.py there. I created a folder in this folder with name "40m" and then I create a folder with name "images" and copied my images that captured by phantom 3 pro in to this folder. then , I opened terminal and I went to OpenDroneMap directory " administrator@administrator-VGN-BZ560P:~/OpenDroneMap$ "   , now , how can I write script to start run.py to procces my iages in 40m folder ?
thank you for attentions

dakotabenjamin, 2016-09-28:
Have you followed the README? https://github.com/OpenDroneMap/OpenDroneMap 

Please tell me where you got stuck.

pahpad, 2016-09-28:
I download the zip file and extract it in home page of ubuntu 15.
then I opened OpenDroneMap folder from home and copied this script in terminal  : 
git clone https://github.com/OpenDroneMap/OpenDroneMap.git .
export PYTHONPATH=$PYTHONPATH:`pwd`/SuperBuild/install/lib/python2.7/dist-packages:`pwd`/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib
bash configure.sh
mkdir build && cd build && cmake .. && make && cd ..
after install , I create a folder by name of "40m" in OpenDroneMap/img folder  then I create a folder and rename it "images" and copied images into it.
for running run.py , how can I write scripte by these directories that I create? 
my images are in folder :   home/OpenDroneMap/img/40m/images
is this script true ?   : python run.py home/OpenDroneMap/img/40m/images   home/OpenDroneMap/img/40m          is it true ?

dakotabenjamin, 2016-09-28:
Try this instead: 
```
python run.py --project-path /home/OpenDroneMap/img/40m
```

pahpad, 2016-09-28:
thank you dear. let me test it.

pahpad, 2016-09-29:
hi dear. I write this script according to my directory on ubuntu  : 
" administrator@administrator-VGN-BZ560P:~/OpenDroneMap/OpenDroneMap$ python run.py --project-path /home/administrator/OpenDroneMap/OpenDroneMap/img/40m "

but a message appeared swiftly : 

Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/administrator/OpenDroneMap/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

then stoped to command line : 

administrator@administrator-VGN-BZ560P:~/OpenDroneMap/OpenDroneMap$ 

please tell me the problem .

pahpad, 2016-09-30:
is there any person to help me solve this problem ?

dakotabenjamin, 2016-10-03:
Check that you set your environment variables correctly as in https://github.com/OpenDroneMap/OpenDroneMap#steps-to-get-opendronemap-running

pahpad, 2016-10-04:
thank you very much for your help.
I unstall ubuntu completely and install ubuntu 16 and use follow you by that link you sent but this message appeared again : 
administrator@pahpad-VGN-BZ560P:~/OpenDroneMap$ python run.py --project-path /home/administrator/OpenDroneMap/img/40m
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/administrator/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory
administrator@pahpad-VGN-BZ560P:~/OpenDroneMap$ 

it seem there is some thing missing . the last line is showing error .
it seems that I should forget linux and opendronemap ;-))))).
please help me if you can. I really like to learn it . 
thank you for your patient and spending time for this error ;-).

dakotabenjamin, 2016-10-04:
OpenDroneMap requires that you set two environment variables before running, `PYTHONPATH` and `LD_LIBRARY_PATH`. This will help python "find" libecto. If you have trouble setting these variables permanently, then there is an alternative run.sh bash script which should set these variables for you:
```
./run.sh --project-path /home/administrator/OpenDroneMap/img/40m
```

Bavar2142, 2016-10-04:
@pahpad How many images are in the dataset? Worst comes to pass you could try node ODM as it has a basic gui tacked on to it. You edit the settings as follows then click add to add your images and then start task to begin processing.

http://nodeodm.masseranolabs.com/

If it fails to run try reducing the default values for min-num-features (integer) and resize-to (integer) by say half.

pahpad, 2016-10-06:
thank you dear for help. you are great friend.
I did this line :   ./run.sh --project-path /home/administrator/OpenDroneMap/img/40m
and ODM started very good and made me very happy , but after some minutes stop by this error:

[INFO]    Running ODM PMVS Cell - Finished
[INFO]    Running ODM Meshing Cell
[DEBUG]   Writing ODM Mesh file in: /home/administrator/OpenDroneMap/img/40m/odm_meshing/odm_mesh.ply
[DEBUG]   running /home/administrator/OpenDroneMap/build/bin/odm_meshing -inputFile /home/administrator/OpenDroneMap/img/40m/pmvs/recon0/models/option-0000.ply -outputFile /home/administrator/OpenDroneMap/img/40m/odm_meshing/odm_mesh.ply -logFile /home/administrator/OpenDroneMap/img/40m/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
sh: 1: /home/administrator/OpenDroneMap/build/bin/odm_meshing: not found
[ERROR]   quitting cause: 
	/home/administrator/OpenDroneMap/build/bin/odm_meshing -inputFile /home/administrator/OpenDroneMap/img/40m/pmvs/recon0/models/option-0000.ply -outputFile /home/administrator/OpenDroneMap/img/40m/odm_meshing/odm_mesh.ply -logFile /home/administrator/OpenDroneMap/img/40m/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
returned with code 32512.

An error occurred. Check stdout above or the logs.
administrator@pahpad-VGN-BZ560P:~/OpenDroneMap$ 

it seems that this error :   sh: 1: /home/administrator/OpenDroneMap/build/bin/odm_meshing: not found      is the problem.
thank you for helping. please help me more to finish this try.
best regards dear.

pahpad, 2016-10-15:
dear dakotabenjamin , I need your help .

dakotabenjamin, 2018-03-28:
This issue was moved to http://community.opendronemap.org/t/odm-path-defining-for-starting-python-run-py-in-terminal-to-process-images/405

closed 2018-03-28 by dakotabenjamin