OpenDroneMap/ODM #388: 'double free or corruption' error during processing.

williamvanderkamp, 2016-09-30:
I ran into the following error while trying to process 612 png drone images on docker.

> 
> [DEBUG]   running /code/SuperBuild/install/bin/texrecon /code/pmvs/recon0/nvmCams.nvm /code/odm_meshing/odm_mesh.ply /code/odm_texturing/odm_textured_model -d gmi -o none     
> /code/SuperBuild/install/bin/texrecon (built on Sep 28 2016, 23:43:13)
> Load and prepare mesh: 
> PLY Loader: comment PCL generated
> Reading PLY: 100025 verts... 199886 faces... done.
> Generating texture views: 
> NVM: Loading file...
> NVM: Number of views: 604
> NVM: Number of features: 0
>         Loading 42%...*** Error in `/code/SuperBuild/install/bin/texrecon': double free or corruption (fasttop): 0x00007fdff8000980 ***
> Aborted (core dumped)
> [ERROR]   quitting cause: 
> 	/code/SuperBuild/install/bin/texrecon /code/pmvs/recon0/nvmCams.nvm /code/odm_meshing/odm_mesh.ply /code/odm_texturing/odm_textured_model -d gmi -o none     
> returned with code 34304.
> 
> An error occurred. Check stdout above or the logs.
> 

I followed precisely the instructions from the final comment on [this issue](https://github.com/OpenDroneMap/OpenDroneMap/issues/369) to run my code.

It seems unlikely this is a problem with my usage, so I thought I would report it here as a bug. I've attached the full log of stdout and stderr (after editing the head and tail to remove my own logging info).

If you'd like to reproduce with the images I used, that could *maybe* be arranged (I'd have to ask my boss).

[edited_outfile.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/503567/edited_outfile.txt)

dakotabenjamin, 2016-10-03:
@alexhagiopol someone else reproduced my error.

dakotabenjamin, 2016-10-04:
So I came across this error when I was testing the pull request for the mvs-texturing feature addition. No one else could reproduce, so I thought it was something wrong on my end. Can you give some information on your run process, your working environment, etc? Was it a clean build?

If you can't share the images publicly, we could arrange for a private exchange. I would be interested in this particularly because this is the first time I've seen this same error.

williamvanderkamp, 2016-10-06:
I had done several runs prior using the same odm_image, but using slightly different sets of docker commands. Some of these runs failed because I was trying to use .tif images, none of which  were detected. After that there is a grey area in my mind, but I know I tried something else with png images, and it didn't work. That prompted me to follow exactly the instructions from the final comment on [this issue](https://github.com/OpenDroneMap/OpenDroneMap/issues/369), rather than my own slightly different set of commands. And then I ran into this bug.

So, it's possible the opendronemap build in odm_image wasn't clean on this final run. I will re-run on a new build of odm_image, when I have the time. I will also preserve the old odm_image entirely, just in case there's something of interest in there. As for sharing the PNGs, I'll ask my boss.

xialang2012, 2016-10-25:
this is a bug for mvs

dakotabenjamin, 2016-10-26:
@xialang2012 do you mean it is a bug in the external MVS-Texturing project? https://github.com/nmoehrle/mvs-texturing

xialang2012, 2016-10-27:
yes, mvs has a bug about import nvm files when converts the images

closed 2020-04-14 by smathermather