OpenDroneMap/ODM #390: odm_extract_utm writes to $PWD

mojodna, 2016-10-04:
https://github.com/OpenDroneMap/OpenDroneMap/blob/540b43d/modules/odm_extract_utm/src/UtmExtractor.cpp#L171

This is awkward if `run.py` is executed from a directory where `$USER` doesn't have write permissions:

```
[DEBUG]   running /code/build/bin/odm_extract_utm -imagesPath /app/projects/seattle_university/images_resize/ -imageListFile /app/projects/seattle_university/opensfm/list_r000.out -outputCoordFile /app/projects/seattle_university/odm_georeferencing/coords.txt -logFile /app/projects/seattle_university/odm_georeferencing/odm_georeferencing_utm_log.txt
sh: 1: cannot create extract_utm_output.txt: Permission denied
[ERROR]   quitting cause:
	/code/build/bin/odm_extract_utm -imagesPath /app/projects/seattle_university/images_resize/ -imageListFile /app/projects/seattle_university/opensfm/list_r000.out -outputCoordFile /app/projects/seattle_university/odm_georeferencing/coords.txt -logFile /app/projects/seattle_university/odm_georeferencing/odm_georeferencing_utm_log.txt
returned with code 256.

An error occurred. Check stdout above or the logs.
```

```
Error in OdmExtractUtm:
Failed to open temporary jhead data file extract_utm_output.txt
For more detailed information, see log file.
```

It should probably have a path specified, similar to the other options.

mbandy, 2017-02-28:
I just ran into this issue as well.

CarlosGrohmann, 2017-05-18:
Just got this same error. Running ODM in a vagrant machine. If I run `run.py` from outside `/odm_app/`, this error happens. Is it creating the file in the directory where it was run from and then moving it to the final place?

dakotabenjamin, 2018-03-28:
jhead has been removed, the code has been changed so this should be fixed.

closed 2018-03-28 by dakotabenjamin