OpenDroneMap/ODM #391: Unable to install OpenDroneMap with a broken packages message

asierrl, 2016-10-05:
I've been trying to install OpenDroneMap following these (https://github.com/OpenDroneMap/OpenDroneMap#steps-to-get-opendronemap-running) instructions, but when running bash configure.sh I get the following message:   
Installing Required Requisites  
Unable to correct problems, you have held broken packages  
ERROR: When Installing Required Requisites.  
Previous lines of the output did not show any error (Updating the system completed smoothly).

I know it seems an Ubuntu issue, but I cannot find any broken package. I've searching the web and most of the solutions I found are very specific to the software they were installing. Also, all the cases were related to the install command...not so sure if the same options apply to bash configure.sh.  
Even though, I've tried a good bunch of proposed solutions:  
* sudo apt-get install -f . Did not do anything (0 updated, 0 newly installed, 0 removed and 589 without updating)  
* The following sequence of commands: 
sudo apt-get update  
sudo apt-get autoclean  
sudo apt-get clean  
sudo apt-get autoremove  
No broken package could be identified in the process.
* The following sequence of commands:
sudo apt-get update --fix-missing
sudo dpkg --configure -a
sudo apt-get install -f
sudo rm /var/lib/apt/lists/lock
sudo rm /var/cache/apt/archives/lock

And some others... Many of the commands wouldn't do anything, nor return any output.
Could someone help me find the broken packages or detect any unmet dependencies?
Thanks a lot.
        Asier

dakotabenjamin, 2016-10-05:
I don't know what specifically you ran so I may post duplicates. Does `dpkg --get-selections | grep hold` return anything? 

Did you try rerunning configure.sh and did you get the same error? It would help if you copy/pasted or posted a screenshot of the terminal what commands you ran.

asierrl, 2016-10-06:
After several hours I had tried too many things to post here, I thought.   
As regards dkpg --get-selections | grep here is the output of the terminal (my Ubuntu language is basque, so that I have translated the output...probably in English Ubuntu doesn't use exactly Grabbed and Unseen, but I think it should be understandable):

asier@asier-lenovo:~$ dpkg --get-selections | grep hold
asier@asier-lenovo:~$ cd OpenDroneMap
asier@asier-lenovo:~/OpenDroneMap$ bash configure.sh
Updating the system
Unseen http://extras.ubuntu.com trusty InRelease
Unseen http://archive.ubuntu.com trusty InRelease                            
Grabbed http://ppa.launchpad.net trusty InRelease                             
Grabbed http://qgis.org trusty InRelease                                      
Grabbed http://extras.ubuntu.com trusty Release.gpg                           
Grabbed http://archive.ubuntu.com trusty-updates InRelease                    
Grabbed http://ppa.launchpad.net trusty InRelease                             
Grabbed http://qgis.org trusty/main Sources                                   
Grabbed http://extras.ubuntu.com trusty Release                               
Grabbed http://archive.ubuntu.com trusty-backports InRelease                  
Unseen http://ppa.launchpad.net trusty InRelease                             
Grabbed http://qgis.org trusty/main i386 Packages                             
Grabbed http://archive.ubuntu.com trusty-security InRelease                   
Grabbed http://extras.ubuntu.com trusty/main Sources                          
Grabbed http://ppa.launchpad.net trusty InRelease                             
Grabbed http://archive.ubuntu.com trusty Release.gpg                          
Grabbed http://extras.ubuntu.com trusty/main i386 Packages                    
Grabbed http://ppa.launchpad.net trusty InRelease                             
Grabbed http://archive.ubuntu.com trusty-updates/restricted Sources           
Grabbed http://ppa.launchpad.net trusty/main i386 Packages                    
Grabbed http://archive.ubuntu.com trusty-updates/universe Sources             
Grabbed http://ppa.launchpad.net trusty/main Translation-en                   
Grabbed http://archive.ubuntu.com trusty-updates/main Sources                 
Grabbed http://ppa.launchpad.net trusty/main i386 Packages                    
Grabbed http://archive.ubuntu.com trusty-updates/multiverse Sources           
Grabbed http://ppa.launchpad.net trusty/main Translation-en                   
Grabbed http://archive.ubuntu.com trusty-updates/main i386 Packages           
Grabbed http://ppa.launchpad.net trusty Release.gpg                           
Grabbed http://archive.ubuntu.com trusty-updates/restricted i386 Packages     
Grabbed http://ppa.launchpad.net trusty/main i386 Packages                    
Grabbed http://archive.ubuntu.com trusty-updates/universe i386 Packages       
Grabbed http://ppa.launchpad.net trusty/main Translation-en                   
Grabbed http://ppa.launchpad.net trusty/main i386 Packages                    
Grabbed http://ppa.launchpad.net trusty/main Translation-en                   
Grabbed http://ppa.launchpad.net trusty Release                               
Grabbed http://ppa.launchpad.net trusty/main i386 Packages                    
Grabbed http://ppa.launchpad.net trusty/main Translation-en                   
Unseen http://extras.ubuntu.com trusty/main Translation-eu_ES                
Unseen http://extras.ubuntu.com trusty/main Translation-eu                   
Unseen http://qgis.org trusty/main Translation-eu_ES                         
Unseen http://extras.ubuntu.com trusty/main Translation-en                   
Unseen http://qgis.org trusty/main Translation-eu                     
Unseen http://qgis.org trusty/main Translation-en                     
Unseen http://dl.google.com stable InRelease         
Grabbed http://dl.google.com stable Release.gpg       
Grabbed http://dl.google.com stable Release
Grabbed http://dl.google.com stable/main i386 Packages
Grabbed http://archive.ubuntu.com trusty-updates/multiverse i386 Packages
Grabbed http://archive.ubuntu.com trusty-updates/main Translation-en          
Grabbed http://archive.ubuntu.com trusty-updates/multiverse Translation-en    
Grabbed http://archive.ubuntu.com trusty-updates/restricted Translation-en    
Grabbed http://archive.ubuntu.com trusty-updates/universe Translation-en      
Unseen https://private-ppa.launchpad.net trusty InRelease                    
Grabbed http://archive.ubuntu.com trusty-backports/main Sources               
Grabbed https://private-ppa.launchpad.net trusty Release.gpg                  
Grabbed http://archive.ubuntu.com trusty-backports/restricted Sources         
Grabbed https://private-ppa.launchpad.net trusty Release                      
Unseen http://dl.google.com stable/main Translation-eu_ES                    
Grabbed http://archive.ubuntu.com trusty-backports/universe Sources           
Grabbed http://archive.ubuntu.com trusty-backports/multiverse Sources         
Grabbed https://private-ppa.launchpad.net trusty/main i386 Packages           
Unseen http://dl.google.com stable/main Translation-eu                       
Grabbed http://archive.ubuntu.com trusty-backports/main i386 Packages         
[... 27 lines cut ...]
Grabbed http://archive.ubuntu.com trusty/multiverse Sources                   
Grabbed http://archive.ubuntu.com trusty/main i386 Packages                   
[... cut ...]
Unseen http://archive.ubuntu.com trusty/restricted Translation-eu_ES         
Unseen http://archive.ubuntu.com trusty/restricted Translation-eu            
Unseen http://archive.ubuntu.com trusty/universe Translation-eu_ES           
Reading the list of packages... Done                                           
Installing Required Requisites
E: Unable to correct problems, you have held broken packages
ERROR: When Installing Required Requisites

Will try other options and upload the terminal's output also. 
Thanks a lot, Benjamin.
Asier

asierrl, 2016-10-07:
Another try:

asier@asier-lenovo:~/OpenDroneMap$ sudo apt-get autoclean
Reading package list... Done
Building dependency tree       
Reading status information... Done
asier@asier-lenovo:~/OpenDroneMap$ sudo apt-get clean
asier@asier-lenovo:~/OpenDroneMap$ sudo apt-get autoremove
Reading package list... Done
Building dependency tree       
Reading status information... Done
0 bertsio berritua(k), 0 berriki instalatuta, 0 kentzeko, eta 589 bertsio-berritu gabe.

Now I try again with bash configure.sh and I get the same result as before.

Thanks again.
Asier

asierrl, 2016-10-07:
And here is the last option:

asier@asier-lenovo:~/OpenDroneMap$ sudo apt-get update
Unseen http://archive.ubuntu.com trusty InRelease                            
Unseen http://extras.ubuntu.com trusty InRelease                             
Grabbed http://qgis.org trusty InRelease                                      
Grabbed http://archive.ubuntu.com trusty-updates InRelease                    
Take:1 http://extras.ubuntu.com trusty Release.gpg [72 B]                     
Grabbed http://qgis.org trusty/main Sources                                   
Grabbed http://archive.ubuntu.com trusty-backports InRelease                  
Grabbed http://extras.ubuntu.com trusty Release                               
Grabbed http://archive.ubuntu.com trusty-security InRelease                   
Grabbed http://qgis.org trusty/main i386 Packages                             
Grabbed http://extras.ubuntu.com trusty/main Sources                          
Grabbed http://archive.ubuntu.com trusty Release.gpg                          
Grabbed http://extras.ubuntu.com trusty/main i386 Packages                    
Grabbed http://archive.ubuntu.com trusty-updates/restricted Sources           
Grabbed http://archive.ubuntu.com trusty-updates/universe Sources             
Grabbed http://archive.ubuntu.com trusty-updates/main Sources                 
Grabbed http://archive.ubuntu.com trusty-updates/multiverse Sources           
Grabbed http://archive.ubuntu.com trusty-updates/main i386 Packages           
Grabbed http://archive.ubuntu.com trusty-updates/restricted i386 Packages     
Grabbed http://archive.ubuntu.com trusty-updates/universe i386 Packages       
Grabbed http://archive.ubuntu.com trusty-updates/multiverse i386 Packages     
Grabbed http://archive.ubuntu.com trusty-updates/main Translation-en          
Grabbed http://archive.ubuntu.com trusty-updates/multiverse Translation-en    
Grabbed http://archive.ubuntu.com trusty-updates/restricted Translation-en    
Grabbed http://archive.ubuntu.com trusty-updates/universe Translation-en      
Grabbed http://archive.ubuntu.com trusty-backports/main Sources               
Unseen http://extras.ubuntu.com trusty/main Translation-eu_ES                
Unseen http://qgis.org trusty/main Translation-eu_ES                         
Unseen http://extras.ubuntu.com trusty/main Translation-eu                   
Unseen http://qgis.org trusty/main Translation-eu                            
Unseen http://extras.ubuntu.com trusty/main Translation-en                   
Grabbed http://archive.ubuntu.com trusty-backports/restricted Sources         
Unseen http://qgis.org trusty/main Translation-en                            
Grabbed http://archive.ubuntu.com trusty-backports/universe Sources           
Grabbed http://archive.ubuntu.com trusty-backports/multiverse Sources         
Grabbed http://archive.ubuntu.com trusty-backports/main i386 Packages         
Grabbed http://archive.ubuntu.com trusty-backports/restricted i386 Packages   
Grabbed http://archive.ubuntu.com trusty-backports/universe i386 Packages     
Grabbed http://archive.ubuntu.com trusty-backports/multiverse i386 Packages   
Grabbed http://archive.ubuntu.com trusty-backports/main Translation-en        
Grabbed http://archive.ubuntu.com trusty-backports/multiverse Translation-en  
Grabbed http://archive.ubuntu.com trusty-backports/restricted Translation-en  
Grabbed http://archive.ubuntu.com trusty-backports/universe Translation-en    
Grabbed http://archive.ubuntu.com trusty-security/restricted Sources          
Grabbed http://archive.ubuntu.com trusty-security/universe Sources            
Grabbed http://archive.ubuntu.com trusty-security/main Sources                
Grabbed http://archive.ubuntu.com trusty-security/multiverse Sources          
Grabbed http://archive.ubuntu.com trusty-security/main i386 Packages          
Grabbed http://archive.ubuntu.com trusty-security/restricted i386 Packages    
Grabbed http://archive.ubuntu.com trusty-security/universe i386 Packages      
Grabbed http://archive.ubuntu.com trusty-security/multiverse i386 Packages    
Grabbed http://archive.ubuntu.com trusty-security/main Translation-en         
Grabbed http://archive.ubuntu.com trusty-security/multiverse Translation-en   
Grabbed http://archive.ubuntu.com trusty-security/restricted Translation-en   
Grabbed http://archive.ubuntu.com trusty-security/universe Translation-en     
Grabbed http://archive.ubuntu.com trusty Release                              
Grabbed http://archive.ubuntu.com trusty/restricted Sources                   
Grabbed http://archive.ubuntu.com trusty/universe Sources                     
Unseen http://dl.google.com stable InRelease                                 
Grabbed http://archive.ubuntu.com trusty/main Sources                         
Grabbed http://ppa.launchpad.net trusty InRelease                             
Grabbed http://archive.ubuntu.com trusty/multiverse Sources                   
Grabbed http://ppa.launchpad.net trusty InRelease                       
Grabbed http://dl.google.com stable Release.gpg                         
Grabbed http://archive.ubuntu.com trusty/main i386 Packages                   
Unseen http://ppa.launchpad.net trusty InRelease                       
Grabbed http://dl.google.com stable Release                             
[... 140 lines cut ...]
Grabbed http://archive.ubuntu.com trusty/multiverse Translation-en            
[... cut ...]
asier@asier-lenovo:~/OpenDroneMap$ sudo apt-get install -f
Reading package list...Done
Building dependency tree     
Reading status information...Done
0 updated,  0 newly installed, 0 removed, eta 589 not updated.
asier@asier-lenovo:~/OpenDroneMap$ sudo rm /var/lib/apt/lists/lock
asier@asier-lenovo:~/OpenDroneMap$ sudo rm /var/cache/apt/archives/lock

Again, trying bash configure.sh gives the same result as before.
Hope you can give me a hint.
I am running an Ubuntu 14.04 32bit on a 64 machine if that may matter.
Thanks a lot.
              Asier

dakotabenjamin, 2016-10-11:
Ok, here is the brute force way to do it, I think.

Manually uninstall all the packages installed by ODM for Required requisites:

```
sudo apt-get remove build-essential \
                     git \
                     cmake \
                     python-pip \
                     libgdal-dev \
                     gdal-bin \
                     libgeotiff-dev \
                     pkg-config
sudo apt-get autoclean
sudo apt-get clean
sudo apt-get autoremove
```
Then manually install each package one-by-one.
```
sudo apt-get install build-essential
sudo apt-get install git
sudo apt-get install cmake
sudo apt-get install python-pip
sudo apt-get install libgdal-dev
sudo apt-get install gdal-bin
sudo apt-get install libgeotiff-dev
sudo apt-get install pkg-config
```

then finally run configure.sh

asierrl, 2016-10-11:
Thanks a lot, Dakota...
Everything went smoothly until I arrived at "sudo apt-get install
libgdal-dev"
There I got this message:

Some packages have been unable to get installed. Perhaps you asked for an
impossible situation, or, if you're using an unstable distribution, some
needed packages have not been produced yet, or continue to be at Incoming.
This information may help fixing the problem:

These packages have unresolved dependencies:
libgdal-dev : Dependency:: libgdal20 (= 2.1.0+dfsg-1~trusty1) but it won't
be installed
                     Dependency:: libgeos-dev but it won't be installed
                     Dependency:: libspatialite-dev but it won't be
installed

E: Unable to fix the problems; you hold broken packages.

At least now we know where the problem is...but I don't understand why
those dependencies cannot be installed...Shall I try to install them
manually?

Thanks a lot!!

mojodna, 2016-10-11:
@asierrl is the ubuntugis PPA configured on your system? (It'll be listed in either `/etc/apt/sources.list` or in a file beneath `/etc/apt/sources.list.d/`) If so, is it `stable` or `unstable`?

Does `sudo apt-get install libgdal20` work, or does that also fail?

If you haven't run `sudo apt-get update` recently, that could also be contributing to the problem.

asierrl, 2016-10-11:
It is the unstable ubuntugis PPA the one configured in my system (but not
deb-src, only deb, if that matters).

I did sudo apt-get update very frequently this week, while trying to fix
the problem. Anyway, I did it again before trying sudo apt-get install
libgdal20, which did not work either:

Some packages have been unable to get installed. Perhaps you asked for an
impossible situation, or, if you're using an unstable distribution, some
needed packages have not been produced yet, or continue to be at Incoming.
This information may help fixing the problem:

These packages have unresolved dependencies:
libgdal20 : Dependency:: libgeos-c1v5 (>= 3.4.2) but it won't be installed

E: Unable to fix the problems; you hold broken packages.

Shall I configure the stable ubuntugis PPA? shall I try to manually install
libgeos-c1v5?
Thanks for your help, Seth.
Best regards,
 Asier

mojodna, 2016-10-11:
Sounds to me like the PPA may have an invalid state at the moment...

You may be best-served at the moment to remove the PPA entirely (assuming that this is a VM or something purely for ODM), since we know that works.

asierrl, 2016-10-11:
I'm afraid it is not...It is my only computer, and I have QGIS installed.
Anyway I disabled unstable ubuntugis, then update and then try again to
install libgdal20. Now it says it cannot find the libgdal20 package.
I tried then to configure ubuntugis stable and then, update and try to
install libgdal20 package again. Same result, it cannot find it.
But then I tried to install libgdal-dev and it worked. I was able then to
follow with the remaining command sequence suggested by Dakota and
configure.sh did work softly.
Now I'll try to run ODM and see what happens.
Thanks a lot to both of you!!!
Asier

asierrl, 2016-10-11:
Oh, no...sorry, I made a mistake... After finishing all the steps suggested
by Dakota Benjamin I run bash configure.sh, which took a lot of time, but
it gave a final error:

1.7/pcl/surface/vtk_smoothing/vtk_mesh_smoothing_laplacian.h

-- Installing:
/home/asier/OpenDroneMap/SuperBuild/install/include/pcl-1.7/pcl/surface/vtk_smoothing/vtk_mesh_smoothing_windowed_sinc.h

[ 97%] Completed 'pcl'

[ 97%] Built target pcl

make: *** [all] Error 2

Script finished

Just in case, I tried to run ODM and got the following:

Traceback (most recent call last):

File "run.py", line 8, in <module>

import ecto

File
"/home/asier/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py",
line 30, in <module>

from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__,
__getitem_slice__, __getitem_str__, __getitem_tuple__, lookup

ImportError: libecto.so.0.6: cannot open shared object file: No such file
or directory

Any hint?
Best regards, and thank you once again.
Asier

mojodna, 2016-10-12:
This sounds like `LD_LIBRARY_PATH` isn't set (it's configured in `run.sh`). If that's the case, you'll also want to make sure that `PYTHONPATH` is set.

dakotabenjamin, 2016-10-12:
Rerun configure.sh and post what you get.

asierrl, 2016-10-12:
You were right, Seth, LD_LIBRARY_PATH and PYTHONPATH were not set. I did it
three days ago (when I started trying) but it seems I did something wrong.
Now the three export instances are within ~/.bashrc, as described in the
wiki.
I have now rerun configure.sh and here is what I got (same error):

Updating the system
Ez ikusi http://dl.google.com stable InRelease
Atzituta http://dl.google.com stable
Release.gpg
Atzituta http://qgis.org trusty
InRelease
Ez ikusi http://archive.ubuntu.com trusty
InRelease
Ez ikusi http://extras.ubuntu.com trusty
InRelease
Atzituta http://dl.google.com stable
Release
Atzituta http://dl.google.com stable/main i386
Packages
Atzituta http://archive.ubuntu.com trusty-updates
InRelease
Atzituta http://extras.ubuntu.com trusty
Release.gpg
Atzituta http://qgis.org trusty/main
Sources
Atzituta http://archive.ubuntu.com trusty-backports
InRelease
Atzituta http://extras.ubuntu.com trusty
Release
Atzituta http://qgis.org trusty/main i386
Packages
Atzituta http://archive.ubuntu.com trusty-security
InRelease
Atzituta http://extras.ubuntu.com trusty/main
Sources
Atzituta http://archive.ubuntu.com trusty
Release.gpg
Atzituta http://extras.ubuntu.com trusty/main i386
Packages
Atzituta http://archive.ubuntu.com trusty-updates/restricted
Sources
Atzituta http://archive.ubuntu.com trusty-updates/universe
Sources
Atzituta http://archive.ubuntu.com trusty-updates/main
Sources
Atzituta http://archive.ubuntu.com trusty-updates/multiverse
Sources
Atzituta http://ppa.launchpad.net trusty
InRelease
Atzituta http://archive.ubuntu.com trusty-updates/main i386
Packages
Atzituta http://ppa.launchpad.net trusty
InRelease
Ez ikusi http://dl.google.com stable/main
Translation-eu_ES
Atzituta http://archive.ubuntu.com trusty-updates/restricted i386
Packages
Ez ikusi http://ppa.launchpad.net trusty
InRelease
Ez ikusi http://dl.google.com stable/main
Translation-eu
Atzituta http://archive.ubuntu.com trusty-updates/universe i386
Packages
Ez ikusi http://dl.google.com stable/main
Translation-en
Atzituta http://ppa.launchpad.net trusty
InRelease
Atzituta http://archive.ubuntu.com trusty-updates/multiverse i386
Packages
[... 1271 lines cut ...]
             ^
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:
In function ‘_vl_kernel_chi2_sse2_f’:
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:414:41:
error: ‘VSIZE’ undeclared (first use in this function)
   T const * X_vec_end = X + dimension - VSIZE ;
                                         ^
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:416:3:
error: unknown type name ‘VTYPE’
   VTYPE vacc = VSTZ() ;
   ^
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:421:7:
error: unknown type name ‘VTYPE’
       VTYPE a = *(VTYPE*)X ;
       ^
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:421:19:
error: ‘VTYPE’ undeclared (first use in this function)
       VTYPE a = *(VTYPE*)X ;
                   ^
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:421:25:
error: expected expression before ‘)’ token
       VTYPE a = *(VTYPE*)X ;
                         ^
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/third_party/vlfeat/vl/mathop_sse2.c:422:13:
erro

dakotabenjamin, 2016-10-12:
#282 seems to be the same issue.

dakotabenjamin, 2016-10-12:
I think this is probably a bug relating to computer architecture. We should investigate to see where compatibility issues exist in OpenSfM w.r.t. 32-bit systems.

asierrl, 2016-10-12:
I tried with the solution posted at
https://github.com/mapillary/OpenSfM/issues/68 and then tried bash
configure.sh again.
There was not any error reported, although I am not sure if that statement
at the end (No install step for 'opensfm') indicates some error.
Here is the output:

asier@asier-lenovo:~/OpenDroneMap$ bash configure.sh
Updating the system
Ez ikusi http://dl.google.com stable InRelease
Atzituta http://dl.google.com stable Release.gpg
Ez ikusi http://archive.ubuntu.com trusty InRelease
Ez ikusi http://extras.ubuntu.com trusty InRelease
Atzituta http://ppa.launchpad.net trusty InRelease
Atzituta http://dl.google.com stable
Release
Atzituta http://extras.ubuntu.com trusty
Release.gpg
Atzituta http://ppa.launchpad.net trusty InRelease
Hartu:1 http://archive.ubuntu.com trusty-updates InRelease [65,9 kB]
Atzituta http://dl.google.com stable/main i386
Packages
Atzituta http://extras.ubuntu.com trusty
Release
Ez ikusi http://ppa.launchpad.net trusty
InRelease
Atzituta http://ppa.launchpad.net trusty
InRelease
Atzituta http://extras.ubuntu.com trusty/main
Sources
Atzituta http://archive.ubuntu.com trusty-backports
InRelease
Atzituta http://ppa.launchpad.net trusty
InRelease
Atzituta http://extras.ubuntu.com trusty/main i386
Packages
Hartu:2 http://archive.ubuntu.com trusty-security InRelease [65,9
kB]
Atzituta http://ppa.launchpad.net trusty
InRelease
Atzituta http://ppa.launchpad.net trusty/main i386
Packages
Atzituta http://archive.ubuntu.com trusty
Release.gpg
Atzituta http://ppa.launchpad.net trusty/main
Translation-en
Hartu:3 http://archive.ubuntu.com trusty-updates/restricted Sources [5360
B]
Atzituta http://ppa.launchpad.net trusty/main i386
Packages
Hartu:4 http://archive.ubuntu.com trusty-updates/universe Sources [166
kB]
Ez ikusi https://private-ppa.launchpad.net trusty
InRelease
Atzituta http://ppa.launchpad.net trusty/main
Translation-en
Ez ikusi http://dl.google.com stable/main
Translation-eu_ES
Atzituta http://ppa.launchpad.net trusty
Release.gpg
Atzituta https://private-ppa.launchpad.net trusty
Release.gpg
Atzituta http://ppa.launchpad.net trusty/main i386
Packages
Ez ikusi http://dl.google.com stable/main
Translation-eu
Atzituta https://private-ppa.launchpad.net trusty
Release
Hartu:5 http://archive.ubuntu.com trusty-updates/main Sources [383
kB]
[... 273 lines cut ...]
[ 76%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/gmm.c.o
[ 79%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/host.c.o
[ 82%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/mathop.c.o
Linking CXX static library libakaze.a
[ 84%] Building C object
CMakeFiles/vl.dir/third_party/vlfeat/vl/getopt_long.c.o
[ 84%] Built target akaze
[ 87%] Building C object
CMakeFiles/vl.dir/third_party/vlfeat/vl/stringop.c.o
[ 89%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/liop.c.o
[ 92%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/sift.c.o
[ 94%] Building C object
CMakeFiles/vl.dir/third_party/vlfeat/vl/rodrigues.c.o
[ 97%] Building C object CMakeFiles/vl.dir/third_party/vlfeat/vl/lbp.c.o
Linking C static library libvl.a
[ 97%] Built target vl
Scanning dependencies of target csfm
[100%] Building CXX object CMakeFiles/csfm.dir/csfm.cc.o
Linking CXX shared library
/home/asier/OpenDroneMap/SuperBuild/src/opensfm/opensfm/csfm.so
[100%] Built target csfm
[ 92%] No install step for 'opensfm'
[ 93%] Completed 'opensfm'
[100%] Built target opensfm
Script finished

dakotabenjamin, 2016-10-12:
Looks like it installed correctly

asierrl, 2016-10-12:
Great!!!
Thanks a lot to both of you.
Best regards,
 Asier

mojodna, 2016-10-12:
@asierrl can you tell us more about your CPU / computer? In addition to generalities, what does `cat /proc/cpuinfo` say?

asierrl, 2016-10-12:
Sure. It is a Lenovo E520 laptop, with a B940 processor and 64
architecture. I have a dual boot install, with Ubuntu 14.04 and windows
7.0, but wih all my data on a third partition (which it's probably nothing
to do with the installation problem, but just in case).
Despite being a 64 bit hardware, my Ubuntu is 32 bit...don't ask me why,
really, I think I simply made a mistake years ago...and now the computer is
close to its end so as to expend to much time changing all the install.
Here is the result of cat /proc/cpuinfo:
asier@asier-lenovo:~$ cat /proc/cpuinfo
processor    : 0
vendor_id    : GenuineIntel
cpu family    : 6
model        : 42
model name    : Intel(R) Pentium(R) CPU B940 @ 2.00GHz
stepping    : 7
microcode    : 0x29
cpu MHz        : 2000.000
cache size    : 2048 KB
physical id    : 0
siblings    : 2
core id        : 0
cpu cores    : 2
apicid        : 0
initial apicid    : 0
fdiv_bug    : no
f00f_bug    : no
coma_bug    : no
fpu        : yes
fpu_exception    : yes
cpuid level    : 13
wp        : yes
flags        : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca
cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx rdtscp lm
constant_tsc arch_perfmon pebs bts xtopology nonstop_tsc aperfmperf
eagerfpu pni pclmulqdq dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm
pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer xsave lahf_lm arat epb
xsaveopt pln pts dtherm
bogomips    : 3991.18
clflush size    : 64
cache_alignment    : 64
address sizes    : 36 bits physical, 48 bits virtual
power management:

processor    : 1
vendor_id    : GenuineIntel
cpu family    : 6
model        : 42
model name    : Intel(R) Pentium(R) CPU B940 @ 2.00GHz
stepping    : 7
microcode    : 0x29
cpu MHz        : 2000.000
cache size    : 2048 KB
physical id    : 0
siblings    : 2
core id        : 1
cpu cores    : 2
apicid        : 2
initial apicid    : 2
fdiv_bug    : no
f00f_bug    : no
coma_bug    : no
fpu        : yes
fpu_exception    : yes
cpuid level    : 13
wp        : yes
flags        : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca
cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe nx rdtscp lm
constant_tsc arch_perfmon pebs bts xtopology nonstop_tsc aperfmperf
eagerfpu pni pclmulqdq dtes64 monitor ds_cpl est tm2 ssse3 cx16 xtpr pdcm
pcid sse4_1 sse4_2 x2apic popcnt tsc_deadline_timer xsave lahf_lm arat epb
xsaveopt pln pts dtherm
bogomips    : 3991.18
clflush size    : 64
cache_alignment    : 64
address sizes    : 36 bits physical, 48 bits virtual
power management:

Hope that helps! Of course, I'll be glad to send any other details you may
need.
By the way. I am running ODM and it seesm to work smoothly!!
Asier

mojodna, 2016-10-12:
That's great, thanks!

I'm inclined to chalk it up to 32-bit-ness, but having your CPU flags to compare with #282 will be helpful.

closed 2020-04-14 by smathermather