OpenDroneMap/ODM #392: problem about the orth module

xialang2012, 2016-10-07:
I have a problem about the ortho result, here is the images used in the test,
http://modis-typechoupload.stor.sinaapp.com/2580320306.zip

and this is the textured result showed in the meshlab, and it is quite good
[image]

but I got an unreasonable ortho result, as shown below
[image]

Does anyone can help me to dig out it, what is wrong about it.

xialang2012, 2016-10-08:
@dakotabenjamin @smathermather

smathermather, 2016-10-09:
Weird. This is very similar to what @pierotofy saw in one of his datasets, see https://github.com/OpenDroneMap/OpenDroneMap/issues/384 . I'm going to close this on the assumption its the same bug, but we can re-open if it's a separate issue.

dakotabenjamin, 2016-10-11:
I think this may be a separate, but related issue. I need more information though.

@xialang2012 can you give more information on the camera used and what manipulations you did on the images? Can you also paste your run command with all the parameters you used?

xialang2012, 2016-10-12:
This images are obtained from the hyper-spectral camera, and I add the geo informtaion, focal length into the EXIF. 

First, I used the command

`python run.py --project-path /home/xl/open/images/odm-1 --matcher-neighbors 4 --min-num-features 8000`

to  run the ODM, but it show me some errors about the resize process.

> [DEBUG]   Resizing dataset to: /home/xl/open/images/odm-1/images_resize
> Traceback (most recent call last):
>   File "run.py", line 36, in <module>
>     plasm.execute(niter=1)
>   File "/home/xl/odm-new/odm-mvs/OpenDroneMap/scripts/resize.py", line 75, in process
>     new_meta['Exif.Photo.PixelXDimension'].value = img_r.shape[0]
>   File "/usr/lib/python2.7/dist-packages/pyexiv2/metadata.py", line 209, in __getitem__
>     return getattr(self, '_get_%s_tag' % family)(key)
>   File "/usr/lib/python2.7/dist-packages/pyexiv2/metadata.py", line 169, in _get_exif_tag
>     _tag = self._image._getExifTag(key)
> KeyError: 'Tag not set'
> xl@ubuntu:~/odm-new/odm-mvs/OpenDroneMap$ 
> 

so I created a folder which named images_resize, and run the command again.

But it seems the UTM module failed, 

> [DEBUG]   running /home/xl/odm-new/odm-mvs/OpenDroneMap/build/bin/odm_extract_utm -imagesPath /home/xl/open/images/odm-1/images_resize/ -imageListFile /home/xl/open/images/odm-1/opensfm/list_r000.out -outputCoordFile /home/xl/open/images/odm-1/odm_georeferencing/coords.txt -logFile /home/xl/open/images/odm-1/odm_georeferencing/odm_georeferencing_utm_log.txt
> Error in OdmExtractUtm:
> No valid GPS position found
> For more detailed information, see log file.
> [ERROR]   quitting cause: 
> 	/home/xl/odm-new/odm-mvs/OpenDroneMap/build/bin/odm_extract_utm -imagesPath /home/xl/open/images/odm-1/images_resize/ -imageListFile /home/xl/open/images/odm-1/opensfm/list_r000.out -outputCoordFile /home/xl/open/images/odm-1/odm_georeferencing/coords.txt -logFile /home/xl/open/images/odm-1/odm_georeferencing/odm_georeferencing_utm_log.txt
> returned with code 256.
> 
> An error occurred. Check stdout above or the logs.

Then I skipped this step and used this command to get the ortho image

`/home/xl/odm-new/odm-mvs/OpenDroneMap/build/bin/odm_orthophoto -inputFile /home/xl/open/images/odm-1/odm_texturing/odm_textured_model.obj -resolution 20 -outputFile 1.jpg`

and I got the ortho image.

dakotabenjamin, 2016-10-12:
Great. I'm running into trouble because I think image 2.jpg somehow got its exif data stripped. No GPS or focal length internally. Can you package and resend?

xialang2012, 2016-10-13:
Yes, you can just remove image 2.jpg , and run it

`python run.py --project-path /home/xl/open/images/odm-1 --matcher-neighbors 4 --min-num-features 8000 --odm_orthophoto-resolution 100`

dakotabenjamin, 2016-10-13:
OK so I ran successfully. Here's what I got:
```
[INFO]    Running ODM Georeferencing Cell
[WARNING] Warning: No coordinates file. Generating coordinates file in: /home/dmb2/opendronemap/xialang/odm_georeferencing/coords.txt
[DEBUG]   running /home/dmb2/opendronemap/build/bin/odm_extract_utm -imagesPath /home/dmb2/opendronemap/xialang/images_resize/ -imageListFile /home/dmb2/opendronemap/xialang/opensfm/list_r000.out -outputCoordFile /home/dmb2/opendronemap/xialang/odm_georeferencing/coords.txt -logFile /home/dmb2/opendronemap/xialang/odm_georeferencing/odm_georeferencing_utm_log.txt
[DEBUG]   running /home/dmb2/opendronemap/build/bin/odm_georef -bundleFile /home/dmb2/opendronemap/xialang/opensfm/bundle_r000.out -inputCoordFile /home/dmb2/opendronemap/xialang/odm_georeferencing/coords.txt -inputFile /home/dmb2/opendronemap/xialang/odm_texturing/odm_textured_model.obj -outputFile /home/dmb2/opendronemap/xialang/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /home/dmb2/opendronemap/xialang/pmvs/recon0/models/option-0000.ply -outputPointCloudFile /home/dmb2/opendronemap/xialang/odm_georeferencing/odm_georeferenced_model.ply -logFile /home/dmb2/opendronemap/xialang/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /home/dmb2/opendronemap/xialang/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   Line: WGS84 UTM 50N

[DEBUG]   lat_frac: 4447447/1 48/1 36360/1000
[DEBUG]   lat_frac: 4447444/1 26/1 5352/1000
[DEBUG]   lat_frac: 4447442/1 13/1 32016/1000
[DEBUG]   lat_frac: 4447443/1 12/1 26820/1000
[DEBUG]   lat_frac: 4447446/1 38/1 38724/1000
[DEBUG]   lat_frac: 4447329/1 1/1 18516/1000
[DEBUG]   lat_frac: 4447445/1 39/1 43884/1000
[DEBUG]   running /home/dmb2/opendronemap/SuperBuild/build/pdal/bin/pdal pipeline -i /home/dmb2/opendronemap/xialang/odm_georeferencing/pipeline.xml --readers.ply.filename=/home/dmb2/opendronemap/xialang/odm_georeferencing/odm_georeferenced_model.ply --writers.las.filename=/home/dmb2/opendronemap/xialang/odm_georeferencing/odm_georeferenced_model.ply.las
[INFO]    Creating geo-referenced CSV file (XYZ format, can be used with GRASS to create DEM)
[INFO]    Running ODM Georeferencing Cell - Finished
[INFO]    Running OMD OrthoPhoto Cell
[DEBUG]   running /home/dmb2/opendronemap/build/bin/odm_orthophoto -inputFile /home/dmb2/opendronemap/xialang/odm_texturing/odm_textured_model_geo.obj -logFile /home/dmb2/opendronemap/xialang/odm_orthophoto/odm_orthophoto_log.txt -outputFile /home/dmb2/opendronemap/xialang/odm_orthophoto/odm_orthophoto.png -resolution 100.0 -outputCornerFile /home/dmb2/opendronemap/xialang/odm_orthophoto/odm_orthophoto_corners.txt
[DEBUG]   Line: WGS84 UTM 50N

[INFO]    Creating GeoTIFF
[DEBUG]   running gdal_translate -a_ullr 452830.418962 4447450.35192 452837.24718 4447442.35565 -co TILED=yes -co COMPRESS=DEFLATE -co PREDICTOR=2 -co BLOCKXSIZE=256 -co BLOCKYSIZE=256 -co INTERLEAVE=band -co NUM_THREADS=ALL_CPUS -a_srs "EPSG:32650" /home/dmb2/opendronemap/xialang/odm_orthophoto/odm_orthophoto.png /home/dmb2/opendronemap/xialang/odm_orthophoto/odm_orthophoto.tif > /home/dmb2/opendronemap/xialang/odm_orthophoto/gdal_translate_log.txt
Warning 6: Driver GTiff does not support NUM_THREADS creation option
[INFO]    Running ODM OrthoPhoto Cell - Finished
[INFO]    OpenDroneMap app finished - Thu Oct 13 08:38:18  2016
dmb2@OpenDroneMap:~/opendronemap$
```

[image]

[image]

For some reason the georeferencing seemed to cause an issue that prevented me from viewing the ortho properly.

dakotabenjamin, 2016-10-13:
This was running off of dev, btw.

xialang2012, 2016-10-14:
Okay it seems that I have an old version. I should update to the last version and then try it again

xialang2012, 2016-10-18:
The problem is the resize parameter.

This comand is ok
`python run.py --project-path /home/xl/open/images/odm-1 --matcher-neighbors 4 --min-num-features 8000 --odm_orthophoto-resolution 100`

But this one failed
`python run.py --project-path /home/xl/open/images/odm-1 --matcher-neighbors 4 --min-num-features 8000 --odm_orthophoto-resolution 100 --resize-to 1000`

Does the ODM require a image size larger than 1000?

xialang2012, 2016-10-19:
I have test another dataset with pixel of 4000*3000, it also failed when I added the `--resize-to 1000` parameter.

dakotabenjamin, 2016-10-19:
What is the minimum threshold for those data? Have you tried 1200?

xialang2012, 2016-10-19:
2200 works well.

xialang2012, 2016-11-02:
Is anyone here, I still can not catch the reason why the ortho-photo module when the image size is less than 2200.

smathermather, 2016-11-02:
@xialang2012 -- just to confirm, you've tried 2000 and 1200?

xialang2012, 2016-11-02:
Yes, I have tried both the 2200, 2000 and 1200, and only the 2200 worked. The original resolution for the images is 1000 * 1000.

dakotabenjamin, 2016-11-02:
I think ODM only resizes to smaller resolutions. what size are the photos in images_resize when you run with 2200? 

Can you paste the logs for the resize step for 2200, 1200, and without resizing?

xialang2012, 2016-11-07:
1. The default one, and the image size in images_resize is 2400 

> xl@ubuntu:~/odm-new/dev/OpenDroneMap$ python run.py --project-path /home/xl/open/images/odm-1/ 
> [INFO]    Initializing OpenDroneMap app - Sun Nov 06 18:14:59  2016
> [INFO]    Running ODM Load Dataset Cell
> [DEBUG]   Loading dataset from: /home/xl/open/images/odm-1/images
> [DEBUG]   Loaded 8.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 3.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 6.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 5.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 1.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 4.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 7.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [INFO]    Found 7 usable images
> [INFO]    Running ODM Load Dataset Cell - Finished
> [INFO]    Running ODM Resize Cell
> [DEBUG]   Resizing dataset to: /home/xl/open/images/odm-1/images_resize
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 8.jpg | dimensions: (2400, 2400, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 3.jpg | dimensions: (2400, 2400, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 6.jpg | dimensions: (2400, 2400, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 5.jpg | dimensions: (2400, 2400, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 1.jpg | dimensions: (2400, 2400, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 4.jpg | dimensions: (2400, 2400, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 7.jpg | dimensions: (2400, 2400, 3)
> [INFO]    Resized 7 images
> [INFO]    Running ODM Resize Cell - Finished
> [INFO]    Running ODM OpenSfM Cell

2. Resize to 1000, and the image size in images_resize is 1000 

> [INFO]    Initializing OpenDroneMap app - Sun Nov 06 18:27:37  2016
> [INFO]    Running ODM Load Dataset Cell
> [DEBUG]   Loading dataset from: /home/xl/open/images/odm-1/images
> [DEBUG]   Loaded 8.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 3.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 6.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 5.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 1.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 4.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 7.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [INFO]    Found 7 usable images
> [INFO]    Running ODM Load Dataset Cell - Finished
> [INFO]    Running ODM Resize Cell
> [DEBUG]   Resizing dataset to: /home/xl/open/images/odm-1/images_resize
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 8.jpg | dimensions: (1000, 1000, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 3.jpg | dimensions: (1000, 1000, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 6.jpg | dimensions: (1000, 1000, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 5.jpg | dimensions: (1000, 1000, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 1.jpg | dimensions: (1000, 1000, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 4.jpg | dimensions: (1000, 1000, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 7.jpg | dimensions: (1000, 1000, 3)
> [INFO]    Resized 7 images
> [INFO]    Running ODM Resize Cell - Finished
> [INFO]    Running ODM OpenSfM Cell

3. Resize to 2200, and the image size in images_resize is 2200 
[... 24 lines cut ...]
> [DEBUG]   Resized 1.jpg | dimensions: (2200, 2200, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 4.jpg | dimensions: (2200, 2200, 3)
> [WARNING] Resize Parameter is greater than the largest side of the image
> [DEBUG]   Resized 7.jpg | dimensions: (2200, 2200, 3)
> [INFO]    Resized 7 images
> [INFO]    Running ODM Resize Cell - Finished
> [INFO]    Running ODM OpenSfM Cell

but resize to 2200 failed to get the correct construction depth map.

> 2016-11-06 18:38:38,179 0 partial reconstructions in total.
> Traceback (most recent call last):
>   File "/home/xl/odm-new/dev/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
>     command.run(args)
>   File "/home/xl/odm-new/dev/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
>     reconstructions = data.load_reconstruction()
>   File "/home/xl/odm-new/dev/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
>     with open(self.__reconstruction_file(filename)) as fin:
> IOError: [Errno 2] No such file or directory: '/home/xl/open/images/odm-1/opensfm/reconstruction.json'
> [ERROR]   quitting cause: 
> 	PYTHONPATH=/home/xl/odm-new/dev/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/xl/odm-new/dev/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/xl/open/images/odm-1/opensfm
> returned with code 256.
> 
> An error occurred. Check stdout above or the logs.

xialang2012, 2016-11-18:
@dakotabenjamin Is something wrong about the process from the logs?

dakotabenjamin, 2016-11-22:
Yeah `[WARNING] Resize Parameter is greater than the largest side of the image` I think we need a smarter resize process

xialang2012, 2016-11-23:
resize to 999, but the result is not reasonable.
[image]

> xl@ubuntu:~/odm-new/dev/OpenDroneMap$ python run.py --project-path /home/xl/open/images/odm-1/ --resize-to 999
> [INFO]    Initializing OpenDroneMap app - Tue Nov 22 17:41:07  2016
> [INFO]    Running ODM Load Dataset Cell
> [DEBUG]   Loading dataset from: /home/xl/open/images/odm-1/images
> [DEBUG]   Loaded 8.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 3.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 6.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 5.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 1.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 4.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [DEBUG]   Loaded 7.jpg | camera:  | dimensions: 1000 x 1000 | focal: 25.0 | ccd: 5.75
> [INFO]    Found 7 usable images
> [INFO]    Running ODM Load Dataset Cell - Finished
> [INFO]    Running ODM Resize Cell
> [DEBUG]   Resizing dataset to: /home/xl/open/images/odm-1/images_resize
> [DEBUG]   Resized 8.jpg | dimensions: (999, 999, 3)
> [DEBUG]   Resized 3.jpg | dimensions: (999, 999, 3)
> [DEBUG]   Resized 6.jpg | dimensions: (999, 999, 3)
> [DEBUG]   Resized 5.jpg | dimensions: (999, 999, 3)
> [DEBUG]   Resized 1.jpg | dimensions: (999, 999, 3)
> [DEBUG]   Resized 4.jpg | dimensions: (999, 999, 3)
> [DEBUG]   Resized 7.jpg | dimensions: (999, 999, 3)
> [INFO]    Resized 7 images
> [INFO]    Running ODM Resize Cell - Finished
> [INFO]    Running ODM OpenSfM Cell

dakotabenjamin, 2016-11-23:
Yea I expect that to happen, I think it's a combination of unexpected results from resize and ODM's poor ability to handle low resolution images. If I had more time to dedicate to this right now I would try to pinpoint the exact problem.

<http://www.clemetparks.com/>

[odm_orthophoto]<https://cloud.githubusercontent.com/assets/14291651/20548804/a4777fa0-b161-11e6-9940-c5284f134184.png>

xialang2012, 2016-11-25:
Waiting for update

smathermather, 2016-11-25:
@xialang2012 -- I don't know that there's any easy solution to images whose resolution is too low. Are these video extracts?

xialang2012, 2016-11-26:
This images come from hyper spectral camera. I think it is the problem of the ortho module, for the textured mesh is nice.

SmithPeng, 2018-08-01:
I've encountered with the same question and hadn't save yet,

SmithPeng, 2018-08-01:
odm_texturing is nice as follow
[image]
orthophoto is odd as follow
[image]

pierotofy, 2018-08-01:
Hey @SmithPeng, are you running the latest version of OpenDroneMap? Would you be able to share the images and the textured model (the `images` and `odm_texturing` folders) to help us troubleshoot it?

xialang2012, 2018-08-05:
It is okay now, so I close the topic.

closed 2018-08-05 by xialang2012