OpenDroneMap/ODM #396: Inexplicably hangs at different points in installation

rcverheggen, 2016-10-14:
I have tried following the vagrant step-by-step several times and have not yet successfully installed the ODM application. It seems to randomly hang at different points throughout the install, and I've never gotten it to hang at the same place twice. I admit I am completely new to command line work, but I am just copying the directions into the git command line window. I wish I could provide a log, but since I have to cut power to get my desktop to respond again, I doubt there is anything logged. I do have a couple screenshots from my phone of the display while it's locked up if would be of any use.  Any help would be appreciated.

Robert

closed 2016-10-20 by rcverheggen

smathermather, 2016-10-27:
:) Glad you figured it out @rcverheggen. Cheers.

smathermather, 2016-10-27:
We should probably both: get some metrics in place for memory / processor ratios and then ultimately have ODM check for likely problems and warn or quit when the base machine configuration is problematic.

Also, adding a sharding approach to break up and re-assemble the dataset is a critical long-term feature, cc @dakotabenjamin @mojodna. I hope to work on wish list stuff today.