OpenDroneMap/ODM #397: No such file or directory: '/.../opensfm/reconstruction.json'

mojodna, 2016-10-14:
After a clean build post-#389, ODM is failing with:

```bash
Traceback (most recent call last):
  File "/code/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/code/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/code/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/code/opensfm/reconstruction.json'
[ERROR]   quitting cause:
    PYTHONPATH=/code/SuperBuild/install/lib/python2.7/dist-packages /code/SuperBuild/src/opensfm/bin/run_all /code/opensfm
returned with code 256.
```

(Paths are originally from within a Docker image, but I was also able to reproduce it in a bare Ubuntu 16.04 instance.)

This occurs when running both `./run.sh --project-path .` and `./run.sh --project-path . --use-opensfm-pointcloud`

mojodna, 2016-10-14:
Ok, so this was partially operator error (due to providing images with bad GPS coordinates), but it would be nice to get a cleaner error message.

In this case, it was that the EXIF data got mangled during the un-distortion process (which may be a problem with CameraCalibration; I'll try to look into that soon):

```diff
-GPS Altitude                    : 134.3 m Above Sea Level
-GPS Latitude                    : 47 deg 30' 30.78" N
-GPS Longitude                   : 121 deg 47' 56.14" W
-GPS Position                    : 47 deg 30' 30.78" N, 121 deg 47' 56.14" W
+GPS Altitude                    : 13438 m Above Sea Level
+GPS Latitude                    : 5262379 deg 2' 4.09" N
+GPS Longitude                   : 590441 deg 40' 24.38" E
+GPS Position                    : 5262379 deg 2' 4.09" N, 590441 deg 40' 24.38" E
```

mojodna, 2016-10-14:
CameraCalibration is fine, yay.

timvand, 2016-11-30:
Getting the same error.

While rerunning same job on a 16 CPU instance, received the same error.  Previously ran the job on a 8 CPU instance without issue.  Prior to running the job on the 16 CPU instance the only directory in place was the images dir.

What gives?

452, 2017-01-26:
> Ok, so this was partially operator error (due to providing images with bad GPS coordinates), but it would be nice to get a cleaner error message.

I Have exactly the same situation

How can I do analyze my geotagged jpg and understand what is wrong with my GPS coordinates?
(I have geotagged images from Mission Planner)

dakotabenjamin, 2017-01-30:
There is a package called [exiftool](http://www.sno.phy.queensu.ca/~phil/exiftool/) which is loaded with features for reading/writing exif. I would look through the manual there. What I've run is something like this:
```
exiftool -a -gps:all /data/images/*
```

just to see all the gps exif data. In your case perhaps the `-if` tag would be useful.

452, 2017-02-02:
I write GPS info manualy with exiftool
```sh
exiftool -overwrite_original sample-data/3/DSC02619.JPG -GPSImgDirection=345.38 -GPSTimeStamp=549455200 -GPSLongitude=30.7651986 -GPSLatitude=50.3296814 -GPSAltitude=232.12 -PitchAngle=-2.08 -RollAngle=6.21
    1 image files updated
```
I have the next GPS info (coordinates exactly like on map)
GPSLatitude: 50 deg 19' 46.85" 
GPSLongitude: 30 deg 45' 54.71"

where can be the problem?

```log
good@ihor:/git/uav$ exiftool -a -gps:all sample-data/3/* -s
======== sample-data/3/DSC02619.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 19' 46.85"
GPSLongitude                    : 30 deg 45' 54.71"
GPSAltitude                     : 232.12 m
GPSTimeStamp                    : 549455200:00:00
GPSImgDirection                 : 345.38
======== sample-data/3/DSC02620.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 19' 49.10"
GPSLongitude                    : 30 deg 45' 55.13"
GPSAltitude                     : 227.8 m
GPSTimeStamp                    : 549459200:00:00
GPSImgDirection                 : 338.78
======== sample-data/3/DSC02621.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 19' 51.31"
GPSLongitude                    : 30 deg 45' 55.27"
GPSAltitude                     : 225.86 m
GPSTimeStamp                    : 549463200:00:00
GPSImgDirection                 : 336.41
======== sample-data/3/DSC02622.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 19' 53.50"
GPSLongitude                    : 30 deg 45' 55.06"
GPSAltitude                     : 228.02 m
GPSTimeStamp                    : 549467200:00:00
GPSImgDirection                 : 322.34
======== sample-data/3/DSC02623.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 19' 55.76"
GPSLongitude                    : 30 deg 45' 54.68"
GPSAltitude                     : 229.08 m
GPSTimeStamp                    : 549471200:00:00
GPSImgDirection                 : 334.53
======== sample-data/3/DSC02624.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 19' 57.94"
GPSLongitude                    : 30 deg 45' 54.69"
GPSAltitude                     : 232.36 m
GPSTimeStamp                    : 549474800:00:00
GPSImgDirection                 : 332.33
======== sample-data/3/DSC02625.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 20' 0.11"
GPSLongitude                    : 30 deg 45' 54.64"
GPSAltitude                     : 233.41 m
GPSTimeStamp                    : 549478600:00:00
GPSImgDirection                 : 328.24
======== sample-data/3/DSC02626.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 20' 2.36"
GPSLongitude                    : 30 deg 45' 54.45"
GPSAltitude                     : 233.83 m
GPSTimeStamp                    : 549482600:00:00
GPSImgDirection                 : 329.85
======== sample-data/3/DSC02627.JPG
[... 649 lines cut ...]
GPSImgDirection                 : 343.93
======== sample-data/3/DSC02720.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 20' 13.13"
GPSLongitude                    : 30 deg 46' 24.83"
GPSAltitude                     : 238.62 m
GPSTimeStamp                    : 550260400:00:00
GPSImgDirection                 : 337.45
======== sample-data/3/DSC02721.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 20' 15.31"
GPSLongitude                    : 30 deg 46' 24.53"
GPSAltitude                     : 239.13 m
GPSTimeStamp                    : 550264200:00:00
GPSImgDirection                 : 341.28
======== sample-data/3/DSC02722.JPG
GPSVersionID                    : 2.3.0.0
GPSLatitude                     : 50 deg 20' 17.54"
GPSLongitude                    : 30 deg 46' 24.24"
GPSAltitude                     : 239.33 m
GPSTimeStamp                    : 550268000:00:00
GPSImgDirection                 : 347.43
    1 directories scanned
  104 image files read
```

dakotabenjamin, 2017-02-06:
So what exact issues are you facing now with ODM? I do not think yours is a GPS issue.

smathermather, 2017-04-22:
I ran into this too. I suspect it's an overly flat data collection, which causes initialization issues in OpenSfM, ala https://github.com/mapillary/OpenSfM/issues/76.

We'll be addressing model initiation in OpenSfM this summer. This might be a good dataset to test. @452 -- is this a dataset you can share for testing?

dakotabenjamin, 2017-12-11:
This will be fixed in #724

kkkarthik6, 2018-02-22:
Hi, I am trying to reconstruct a scene and it has good feature matches and all but is giving me following error

2018-02-22 11:24:35,152 DEBUG: Robust matching time : 0.000420093536377s
2018-02-22 11:24:35,152 DEBUG: Full matching 125 / 146, time: 0.129931926727s
2018-02-22 11:24:35,221 DEBUG: 04.jpg - 05.jpg has 262 candidate matches
2018-02-22 11:24:35,222 DEBUG: Robust matching time : 0.000317096710205s
2018-02-22 11:24:35,222 DEBUG: Full matching 237 / 262, time: 0.123347043991s
2018-02-22 11:24:35,695 INFO: reading features
2018-02-22 11:24:35,742 DEBUG: Merging features onto tracks
2018-02-22 11:24:35,780 DEBUG: Good tracks: 1068
2018-02-22 11:24:36,181 INFO: Starting incremental reconstruction
2018-02-22 11:24:37,473 INFO: 0 partial reconstructions in total.
Traceback (most recent call last):
File "/home/karthik/spring/OpenSFM/OpenSfM/bin/opensfm", line 34, in
command.run(args)
File "/home/karthik/spring/OpenSFM/OpenSfM/opensfm/commands/mesh.py", line 22, in run
reconstructions = data.load_reconstruction()
File "/home/karthik/spring/OpenSFM/OpenSfM/opensfm/dataset.py", line 371, in load_reconstruction
with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: 'data/IR2/reconstruction.json'

Kindly let me know what issue can this be.

smathermather, 2018-03-16:
Can you tell me what your camera is? This is a common issue with cameras with barrel distortion that hasn't been corrected for.

closed 2018-03-28 by dakotabenjamin