OpenDroneMap/ODM #398: Odd orthophoto but good point cloud

asierrl, 2016-10-14:
I'm experiencing a problem similar to that of #392, although the origin doesn't seem the same, as I have no gps info in the exif and rather I run ODM with a GCP file.
When running with default parameters I run out of memory, and hence I tried again with resize-image=1200. My orthophoto looks like this:
[image]

But, my point cloud looks nice:
[image]

Odd enough...I run this same set of images with resize=1200 one year ago (different computer, I was trying to check now that everything installed correctly) and got good results:

[image]

Both orthos have been copied from QGIS, with the same view extent. So, appart from the interative mosaic-like patter, there is also a size or georeferencing  problem.
I used exactly the same files as one year before (including the gcp file).
Any hint?
Thanks a lot!!
Asier

dakotabenjamin, 2016-10-17:
@asierrl recently I pushed some changes in #386 which changed which images to pull from when georeferencing. Do you think this could be the cause?

asierrl, 2016-10-17:
Hi, Benjamin....Honestly...I don't have a clue if that might be the
problem. Sorry, I'm too ignorant on these issues. But I am glad to try to
guess.
I installed ODM last week, so that if you did that change 18 days ago, the
change should be already within my ODM version. But I saw you did this
change in OpenDroneMap:dev. I read several times that ODM can be run from
different branches...and dev is one of them. Perhaps I could try that, but
I don't know how to do it. Currently I go into the OpenDroneMap folder and
run the "python run.py..." command. How could I do to run it on dev?
Thanks!!
Asier

dakotabenjamin, 2016-10-17:
```
git checkout dev
bash configure.sh
python run.py ...
``` 

should do

dakotabenjamin, 2016-10-17:
That being said, I am now unconvinced of my original theory.

asierrl, 2016-10-19:
Mmm...yeah, I guess it was a memory issue. In fact, I did as you suggested, running it in dev branch and run out of memory. I already did a resize-to 1200. I don't know if I can go lower, but I probably will give it a try and let you know.
Thanks for your support and help (and for the software :-) )
Best regards,
Asier

asierrl, 2016-10-19:
Ah...and I forgot... I always get a warning that ccd width was not found for my camera. Do you think that could be the reason for the odd orthophoto? I would say no, but for no good reason.
I made a pull request to include it in sensor_data.json. It was included and it worked (same data I'm trying now). I have checked and in fact the camera model is included, but it says Panasonic Lumix DMC-GH4. Instead exif info refers to it as panasonic dmc-gh4. Should I include it as a new camera in sensor_data.json? should I forced a ccwidth within the command? or is it a way for ODM to recognize it as the same camera?
Best regards, 
Asier

xialang2012, 2016-10-24:
The Orthophoto module has some problems to co-work with mvs-texturing.

asierrl, 2016-10-25:
So, any idea on how to avoid this effect? or about its relationship with
the amount of memory?

dakotabenjamin, 2016-10-26:
How much memory do you have?

asierrl, 2016-10-26:
I got last year's good result on my desktop, with i7, 32Gb RAM and a 3Gb
graphic card.
Now I am running it on my laptop, with a Intel B940 processor, 8 Gb RAM and
an integrated Intel HD graphic card (1696 Mb graphic memory available, 64
Mb dedicated to video, 1632 Mb shared with the system).

asierrl, 2016-10-26:
8 Gb RAM