OpenDroneMap/ODM #401: Not able to have geo data on orthophotos

GCorbel, 2016-10-18:
Hi,

I created orthophotos with [those images](https://github.com/OpenDroneMap/odm_data_caliterra/tree/b797a3a07dbde119d537d38b124918c9455c5cee). It works but when I do `exiftool odm_orthophoto/odm_orthophoto.tif`, I have this exif : 

```
ExifTool Version Number         : 9.46
File Name                       : odm_orthophoto.tif
Directory                       : odm_orthophoto
File Size                       : 393 MB
File Modification Date/Time     : 2016:10:18 14:08:31-04:00
File Access Date/Time           : 2016:10:18 14:09:22-04:00
File Inode Change Date/Time     : 2016:10:18 14:08:31-04:00
File Permissions                : rw-r--r--
File Type                       : TIFF
MIME Type                       : image/tiff
Exif Byte Order                 : Little-endian (Intel, II)
Image Width                     : 10148
Image Height                    : 10152
Bits Per Sample                 : 8 8 8 8
Compression                     : Uncompressed
Photometric Interpretation      : RGB
Strip Offsets                   : (Binary data 98788 bytes, use -b option to extract)
Samples Per Pixel               : 4
Rows Per Strip                  : 1
Strip Byte Counts               : (Binary data 60911 bytes, use -b option to extract)
Planar Configuration            : Chunky
Extra Samples                   : Unassociated Alpha
Sample Format                   : Unsigned; Unsigned; Unsigned; Unsigned
Pixel Scale                     : 0.0499957351202189 0.0499996109140877 0
Model Tie Point                 : 0 0 0 440680.854065 4564323.03883 0
Geo Tiff Version                : 1.1.0
GT Model Type                   : Projected
GT Raster Type                  : Pixel Is Area
GT Citation                     : WGS 84 / UTM zone 17N
Geog Citation                   : WGS 84
Geog Angular Units              : Angular Degree
Projected CS Type               : WGS84 UTM zone 17N
Proj Linear Units               : Linear Meter
Image Size                      : 10148x10152
```

There is no information about geolocation. What can I do ?

I'm using the docker version and the master branch.

Thanks !

mojodna, 2016-10-18:
GeoTIFFs include location information differently. To see geographic information, use `gdalinfo`:

```
gdalinfo odm_orthophoto.tif
```

GCorbel, 2016-10-18:
Thanks for your help. I have this result : 

```
Driver: GTiff/GeoTIFF
Files: odm_orthophoto/odm_orthophoto.tif
Size is 10148, 10152
Coordinate System is:
PROJCS["WGS 84 / UTM zone 17N",
    GEOGCS["WGS 84",
        DATUM["WGS_1984",
            SPHEROID["WGS 84",6378137,298.257223563,
                AUTHORITY["EPSG","7030"]],
            AUTHORITY["EPSG","6326"]],
        PRIMEM["Greenwich",0,
            AUTHORITY["EPSG","8901"]],
        UNIT["degree",0.0174532925199433,
            AUTHORITY["EPSG","9122"]],
        AUTHORITY["EPSG","4326"]],
    PROJECTION["Transverse_Mercator"],
    PARAMETER["latitude_of_origin",0],
    PARAMETER["central_meridian",-81],
    PARAMETER["scale_factor",0.9996],
    PARAMETER["false_easting",500000],
    PARAMETER["false_northing",0],
    UNIT["metre",1,
        AUTHORITY["EPSG","9001"]],
    AXIS["Easting",EAST],
    AXIS["Northing",NORTH],
    AUTHORITY["EPSG","32617"]]
Origin = (440680.854065000021365,4564323.038829999975860)
Pixel Size = (0.049995735120219,-0.049999610914088)
Metadata:
  AREA_OR_POINT=Area
Image Structure Metadata:
  INTERLEAVE=PIXEL
Corner Coordinates:
Upper Left  (  440680.854, 4564323.039) ( 81d42'27.98"W, 41d13'41.25"N)
Lower Left  (  440680.854, 4563815.443) ( 81d42'27.80"W, 41d13'24.79"N)
Upper Right (  441188.211, 4564323.039) ( 81d42' 6.19"W, 41d13'41.38"N)
Lower Right (  441188.211, 4563815.443) ( 81d42' 6.01"W, 41d13'24.92"N)
Center      (  440934.532, 4564069.241) ( 81d42'16.99"W, 41d13'33.09"N)
Band 1 Block=10148x1 Type=Byte, ColorInterp=Red
  Mask Flags: PER_DATASET ALPHA
Band 2 Block=10148x1 Type=Byte, ColorInterp=Green
  Mask Flags: PER_DATASET ALPHA
Band 3 Block=10148x1 Type=Byte, ColorInterp=Blue
  Mask Flags: PER_DATASET ALPHA
Band 4 Block=10148x1 Type=Byte, ColorInterp=Alpha
```

I'm not very familiar with geomatic but I'm not sure it makes sense. Can you confirm ? If results are correct, how to have normal gps positions ?

mojodna, 2016-10-18:
Yup, that makes sense. Resulting GeoTIFFs are [projected](https://en.wikipedia.org/wiki/Map_projection) to [UTM](https://en.wikipedia.org/wiki/Universal_Transverse_Mercator_coordinate_system) coordinates, in this code, zone 17N. GDAL includes the corner coordinates in lng/lat, but as degrees/minutes/seconds: `81d42'16.99"W, 41d13'33.09"N`

You can do the conversion using `gdaltransform`. This converts the center coordinate from UTM zone 17N (`EPSG:32717`) to lng/lat (WGS 84 / `EPSG:4326`):

```bash
$ echo "440934.532 4564069.241" | gdaltransform -s_srs EPSG:32617 -t_srs EPSG:4326
-81.7047201997069 41.225857199375 0
```

Here's a useful presentation from Maptime Boston / Maptime DC about projections and coordinate systems: http://briandaviddavidson.com/projections/#/

GCorbel, 2016-10-18:
Nice! Thanks !

pierotofy, 2016-10-19:
@GCorbel also check Chapter 6 from PostGIS in Action 2nd Edition for a great overview on the subject: "Spatial reference system considerations".

closed 2016-10-24 by GCorbel

smathermather, 2016-10-27:
Just to add clarification -- normal TIFF geotags only specify a single point, generally the location of the camera. GeoTIFFs effectively provide coordinates for all pixels in the image.