OpenDroneMap/ODM #402: DJI Phantom 4 Images

hugowikstrom, 2016-10-20:
Hello

I really like ODM, but I have problems to process the images from my DJI Phantom 4. It seems to be missing some information in the images .. Is there any way to prepare the images so that they work in ODM?

Anyone have an idea?

pierotofy, 2016-10-20:
What information seems to be missing? Could you copy the output of ODM?

hugowikstrom, 2016-10-22:
hugo@Asus-K72Dr:~/odm/OpenDroneMap$ ./run.sh --project-path ../vartan/
[INFO]    Initializing OpenDroneMap app - Sat Oct 22 08:38:17  2016
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /home/hugo/odm/vartan/images
[WARNING] Could not find ccd_width in file. Use --force-ccd or edit the sensor_data.json file to manually input ccd width
[DEBUG]   Loaded DJI_0594.JPG | camera: dji fc330 | dimensions: 4000 x 3000 | focal: 3.61 | ccd: None
Premature end of JPEG file
Traceback (most recent call last):
  File "/home/hugo/odm/OpenDroneMap/run.py", line 38, in <module>
    plasm.execute(niter=1)
  File "/home/hugo/odm/OpenDroneMap/scripts/dataset.py", line 57, in process
    self.params.force_ccd)
  File "/home/hugo/odm/OpenDroneMap/opendm/types.py", line 32, in __init__
    self.parse_pyexiv2_values(self.path_file, force_focal, force_ccd)
  File "/home/hugo/odm/OpenDroneMap/opendm/types.py", line 77, in parse_pyexiv2_values
    self.width = img.shape[1]
AttributeError: 'NoneType' object has no attribute 'shape'

hugowikstrom, 2016-10-22:
I also test with option --force-ccd but it did not work.

pierotofy, 2016-10-22:
I think the problem is `Premature end of JPEG file`, it has nothing to do with ccd. It looks like DJI_0594.JPG might be corrupted? Could you attach the file here?

hugowikstrom, 2016-10-22:
[image]

pierotofy, 2016-10-22:
Image looks fine; it might have to do with the way you built opendronemap. Did you build from the `dev` branch?

pierotofy, 2016-10-22:
Processing for that image works fine here: http://nodeodm.masseranolabs.com/ (it runs the latest `dev` code).

hugowikstrom, 2016-10-22:
You seem to be on track that the image is incorrect. I run a different set of images now and it seems to work better. I will return.

I compiled the entire ODM last weekend .. How do I install the dev version

pierotofy, 2016-10-22:
```
git clone https://github.com/OpenDroneMap/OpenDroneMap.git
git checkout dev
```

Then follow the readme: https://github.com/OpenDroneMap/OpenDroneMap/tree/dev