OpenDroneMap/ODM #403: reversed image

rcverheggen, 2016-10-20:
I ran a project through ODM and got an interesting result - the image was reversed.  I am not sure how accurate the georeferencing is, but i do know that ESRI's Drone2Map dropped it in the right place.  
Below is a side-by-side of a bing maps image (left) and the ODM image (right).
[image]

any thoughts or ideas on how to fix this?

I ran another image set through it yesterday, and it worked perfectly.
Thanks in advance

mojodna, 2016-10-21:
Are there other files in the source `images/` directory? (E.g. non-JPEGs)

rcverheggen, 2016-10-21:
There are no other files in the images folder.

xialang2012, 2016-10-24:
We need to reverse it? I also face the same problem about it.

rcverheggen, 2016-10-25:
Any thoughts on this?  I can make the image folder available somewhere if you think that would help diagnose the issue.

dakotabenjamin, 2016-10-26:
How do you have your photos georeferenced? If there's no gps data then the software won't know which way is up

rcverheggen, 2016-10-26:
Well, I took the photos straight off the memory card from a phantom 3.  I have run other sets of photos from the same UAV in the same way through ODM and they come within a few feet (and not reversed).  The unit had GPS connectivity through both runs I have tried in ODM.

piousgreek, 2017-02-07:
Sorry to dig up an older thread or commandeer it, but i am running into this issue as well.  The image is flipped.  I am using images that are geotagged in mission planner and i believe them to have accurate location data.  The coords.txt file indicates WGS84 UTM 11N which is accurate for my location in Fresno, California.

How do I tell ODM which side is up?

closed 2018-03-28 by dakotabenjamin