OpenDroneMap/ODM #406: Input photo coordinates are written as UTM

mojodna, 2016-10-24:
In [`odm_georeferencing.py`](https://github.com/OpenDroneMap/OpenDroneMap/blob/dev/scripts/odm_georeferencing.py#L120-L121), input files have their coordinates rewritten as UTM within the EXIF data.

[IMO] input files shouldn't be touched during processing, particularly as the pipeline can only handle input files with Lng/Lat coordinates.

This is the root cause of #397, as I was re-processing imagery that had already been run through.

mojodna, 2016-10-24:
(Also volunteering to fix this one, though it's an easy removal, so if someone else gets to it first, that's cool ;-)

dakotabenjamin, 2016-10-26:
iirc the input photos are the resized photos but i'll have to double-check

closed 2018-03-28 by dakotabenjamin