OpenDroneMap/ODM #408: Remove white around the orthophoto

GCorbel, 2016-10-24:
Hi,

When I generate an Orthophoto, there is always an ugly border like you can see here : 
[image] . Is it possible to generate a cleaner image ?

Thanks !

elRubas, 2016-10-24:
in my experience, all the codes I had used give you in the result the border and if you don't needed you can removed manually in programs like meshlab

Bavar2142, 2016-10-24:
@elRubas Is there a chance you could tell us how?

GCorbel, 2016-10-24:
Thanks for anwser @elRubas I cannot use meshlab because I'm trying to automate a process. Any other idea ?

elRubas, 2016-10-24:
@Bavar2142  here is a youtube tutorial video how you can do, Its more easy and fast see than explain

https://www.youtube.com/watch?v=m2nmeJj5Ij4

@GCorbel  I really don't have idea how to give you automaticly a clean resut but if I see how to do it I will give you the info

dakotabenjamin, 2016-10-26:
The white background should be transparent. Can you try opening the png in something like GIMP to see if it's not the software you are using?

GCorbel, 2016-10-26:
I have the same result with GIMP.

GCorbel, 2016-10-31:
It works with the `convert` command of image magick. Here is an example `convert source.png -fuzz 1% -transparent white target.png`.

closed 2016-10-31 by GCorbel