OpenDroneMap/ODM #410: Package 'libvtk5-dev' has no installation candidate on bash configure

Bavar2142, 2016-10-25:
E: Package 'libvtk5-dev' has no installation candidate
ERROR: Error when Installing Dependencies Requisites

Hello all,
had to reinstall ubuntu on the vm so i had to redo the odm install for 16.10. I ran into the above error and tried to find it via synaptic package manager with no luck.
Does anyone have an idea?

dakotabenjamin, 2016-10-26:
https://launchpad.net/ubuntu/yakkety/+source/vtk

It looks like no release yet. @smathermather

Bavar2142, 2016-10-26:
Ah, that would explain it. I tried with libvtk6 with no luck either.

smathermather, 2016-10-27:
Is this a blocking issue for 16.04 as well?

mojodna, 2016-10-27:
No. I've been running on 16.04 with no problems.

dakotabenjamin, 2016-10-27:
No, it is only for 16.10. Should go away soon.

[http://sig.cmparks.net/cmp-ms-90x122.png]Dakota Benjamin
Geospatial Developer
clevelandmetroparks.com<http://www.clemetparks.com/>

<http://www.clemetparks.com/>

Bavar2142, 2016-10-27:
@dakotabenjamin did you have an idea of when thatd be?

dakotabenjamin, 2016-10-28:
No, we don't have any control over these dependencies. I'm assuming it won't be too long because this looks like a well-maintained package.

dakotabenjamin, 2016-10-28:
https://answers.launchpad.net/ubuntu/+source/vtk/+question/403584

The package has been replaced. Reopening so that we can get this fixed

Bavar2142, 2016-10-30:
Good to know its being looked at.

jpmaniac, 2016-12-16:
Any luck on this? I'm on 16.10 and am unable to configure ODM still on the latest Release Candidate

dakotabenjamin, 2016-12-19:
This problem is more complicated than I thought. We have some older code that vtk6 appears to break

dakotabenjamin, 2016-12-19:
I don't know how much legacy code still exists in some of our modules, but it looks like updating from vtk5 to vtk6 breaks all of them. PMVS and odm_meshing most notably. I do not have the c++ skills to get dirty with the code to port it over, but I will put this on our priorities list. 

The best solution for now is to use [Docker to containerize ODM](https://github.com/OpenDroneMap/OpenDroneMap#build-and-run-using-docker).

smathermather, 2016-12-21:
Should we get rid of PMVS? I realize that only solves half the problem, but at least it's a start.

Created an issue for PMVS: https://github.com/OpenDroneMap/OpenDroneMap/issues/442

dakotabenjamin, 2017-09-01:
#659 is related.

closed 2018-03-28 by dakotabenjamin