OpenDroneMap/ODM #414: Modules not found

rx099138, 2016-11-01:
Hello - This is my first try with OpenDroneMap and having some problems getting setup. I have followed all instructions on the readme and still getting problems like this:

[root@ip-10-95-177-162 OpenDroneMap]# python run.py --project-path /home/ec2-user/corbet_glen/
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
ImportError: No module named ecto

I am using AWS and have tried multiple versions of Ubuntu, the AWS Linux AMI, and both dev and branch code for OpenDroneMap. All combinations give different errors. The build is successful, but the software won't run.

Can anyone send instructions on how to get this going?

dakotabenjamin, 2016-11-01:
It looks like you haven't set the environment variables: try switching `python run.py` to `./run.sh` (keeping everything else the same) and it will set the variables for you.

rx099138, 2016-11-01:
Thank You. Running as root I get permissions problems. Checking into that now.
[root@ip-10-95-177-162 OpenDroneMap]# ./run.py --project-path /home/ec2-user/corbet_glen/
bash: ./run.py: Permission denied

dakotabenjamin, 2016-11-01:
Note you should be running the bash script `run.sh`

rx099138, 2016-11-01:
Ok. Tried that with the same outcome.

[root@ip-10-95-177-162 OpenDroneMap]# ./run.sh --project-path /home/ec2-user/corbet_glen/
Traceback (most recent call last):
  File "/home/ec2-user/OpenDroneMap/run.py", line 8, in <module>
    import ecto
ImportError: No module named ecto

dakotabenjamin, 2016-11-01:
In that case it looks like your configuration is incorrect. Please rerun configure.sh and paste the output here.

rx099138, 2016-11-01:
Thank You. Now I get this...so I'll have to build a new Ubuntu box. This one happens to be running AWS Linux.

Be back here soon to advise results.
ERROR: Not a Debian-based linux system. 
           Impossible to install OpenCV with this script

dakotabenjamin, 2016-11-01:
Oh, right. Sorry I should have caught that earlier.

[http://sig.cmparks.net/cmp-ms-90x122.png]Dakota Benjamin
Geospatial Developer
clevelandmetroparks.com<http://www.clemetparks.com/>

<http://www.clemetparks.com/>

rx099138, 2016-11-02:
@dakotabenjamin Thank You for your help today. I built a new Ubuntu machine in AWS and used a different AMI. After a successful build, my first OpenDroneMap is processing now! Thanks again.

closed 2016-11-02 by rx099138