OpenDroneMap/ODM #415: how can i compile orthophoto module independently

assbben, 2016-11-03:
here are my command in dir modules/odm_orthophoto
mkdir build
cd build
cmake ..
make 
but it has some errors
`error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’`
`error: no matching function for call to ‘std::vector<std::vector<Eigen::Matrix<float, 2, 1>, Eigen::aligned_allocator<Eigen::Matrix<float, 2, 1> > > >::push_back(std::vector<Eigen::Matrix<float, 2, 1> >&)’`

therefore i used make -i to compile it
although it said `[100%] Built target odm_orthophoto` i cant found it

or there has another way to compile it

thanks

bryanibit, 2016-11-21:
To be honest, I want to compile orthophoto module independly too.Recently I found that orthophoto is as a cell of ecto like other module. Specificlly, I still not understand the code of ODM. I think you could start to see the script/orthophoto.py and try to understand the orthophoto module.

juvinski, 2017-07-08:
To solve this issue I had to use the Eigen 3.2.10, with this version, PCL 1.8.0 and VTK 7.1.1 - compiling and installing manually(downloading and compiling).
Btw, I' m using Ubuntu 17.04.

dakotabenjamin, 2018-03-28:
I know this is old, but you can find the orthophoto module in `build/bin`

closed 2018-03-28 by dakotabenjamin