OpenDroneMap/ODM #416: Document testing better

mairsbw, 2016-11-05:
Right now tests exist in `/tests`, but it's unclear how to run them. The Markdown file in that folder I think is also a list of missing tests, so that should be clarified as well. I'd suggest adding these to the README.

If someone can clarify the procedure here, I can add it to the README and submit a PR.

pierotofy, 2016-11-09:
I wonder if an entry should be added in the Wiki instead. I'll wait for @smathermather to comment on this also.

mairsbw, 2016-11-09:
The problem with the wiki, which is actually a problem with this thing right now, is that the wiki isn't per-branch. So right now all links to the wiki are broken in the README on master because they have been changed but only fixed on dev.

pierotofy, 2016-11-09:
I know that a merge of the dev branch into master should be coming soon. That will probably fix the links problem.

smathermather, 2017-01-18:
I know one thing that @dakotabenjamin is thinking about long term is proper documentation frameworks as opposed to the wiki. Any thoughts or ideas welcome of course.

dakotabenjamin, 2018-03-28:
Closing this in favor of #516

closed 2018-03-28 by dakotabenjamin