OpenDroneMap/ODM #417: Document running an example

mairsbw, 2016-11-05:
There are a variety of example datasets in the https://github.com/OpenDroneMap/odm_data project. It'd be great if that was documented in the README as a suggestion for learning how to use OpenDroneMap. Having a sample set of imagery along with the example results (maybe just an orthomosaic) would help some people understand the workflow a lot more I think.

I'm capable of writing this up now, though I'm getting a bad orthomosaic for one of the image sets. Pointing people to the dataset likely to give the best results would be ideal.

dakotabenjamin, 2016-11-23:
I think the toledo dataset is a great one to use, it generally works well. Improving the documentation is one of my priorities (when I can finally find the time)

mairsbw, 2016-11-23:
Well I'd be happy to help, especially since I'm basically learning all these things for the first time and can point out deficiencies, improvements, and I'm even a decent writer. So let me know.

Bavar2142, 2016-11-23:
I'd be up for testing any examples that get made once I reinstall ubuntu 15.10.

pierotofy, 2016-11-26:
@mairsbw I'd be great to get help to improve the docs. Perhaps you can make a fork of the documents you want to improve, make a few changes, then make a PR and we can review / give you feedback. Or are you looking for more specific guidance?

dakotabenjamin, 2016-11-30:
@mairsbw I've created a [documentation update branch](https://github.com/OpenDroneMap/OpenDroneMap/tree/documentation-update) and there's a spot in the README.md for an example. If I don't get around to it this week feel free to contribute. 

I recommend using the [odm_data_aukerman](https://github.com/OpenDroneMap/odm_data_aukerman) dataset because it's already mentioned in the quickstart.

dakotabenjamin, 2016-11-30:
The biggest concern i have with adding example data is we must make sure the image file sizes are very compact so they don't add to the already weighty repository. It may be better to simply link to the wiki and add an example page there.

mairsbw, 2016-11-30:
We don't need to add data, the data is already in the odm_data repos. My
proposal is that we document all of those datasets (I already filed an
issue in one of them, but there's been no movement there) and that we
choose 1 or 2 datasets as canonical examples showing off some of the
functionality (maybe 1 is an agriculture example, another is a 3d model of
a construction/mining site, etc.).

Best regards,

Bryant Mairs

Integrations Engineer | bryant@intelinair.com <email@intelinAir.com>

612-309-5455 | www.intelinair.com

San Jose, California

@Intelinair | linkedIn.com/intelinair

mairsbw, 2016-12-05:
So I've already documented the Seneca dataset: https://github.com/OpenDroneMap/odm_data_seneca

dakotabenjamin, 2016-12-05:
Nice! Looks good. 

I would recommend rerunning seneca with the `--mvs_texturing-skipGlobalSeamLeveling` tag to reduce that gradient.

mairsbw, 2016-12-05:
I tried doing that and I didn't get any better results, but I'll double-check when I have time. Additionally I'll try to go through the rest of the datasets as I have time as these are definitely useful to the wider community as well.

dakotabenjamin, 2016-12-05:
I will work on updating the repository with the new (better) datasets. I think I need to update the boruszyn_kap submodule as well with new GCPs.

mairsbw, 2016-12-05:
I don't know exactly what that means. Should I continue adding READMEs for the other datasets as far as I can or how can I help here?

dakotabenjamin, 2016-12-05:
I'm going to clean up and update the odm_data repository to contain links to more datasets.

mairsbw, 2016-12-05:
Sounds good. We should track that issue somewhere else if you want to collaborate on it.

As for this issue, it was really about choosing some datasets that highlight some of the features of ODM and documenting them in the README. Should I focus on the odm_data_aukerman dataset and get it documented in this README for this? We can expand to additional datasets in a future tracking issue?

dakotabenjamin, 2016-12-05:
Yeah that sounds good to me.

closed 2018-03-28 by dakotabenjamin