OpenDroneMap/ODM #418: Help needed - Poor results from multiple datasets

aquilaimaging, 2016-11-06:
Hello

Hoping someone smarter than I could tell me what I did wrong with these two missions detailed below. Let me know if you need additional details. Thanks in advance for your help :)

-------------------------------------------------------
**Here's links to the two datasets and logs from the flights:** 

SX260:
https://drive.google.com/open?id=0B7HD5XkwsmeJLWZtcTBBSmZ5ZzQ

PowerShot N: 
https://drive.google.com/open?id=0B7HD5XkwsmeJdGh6cUQ3Vnl2ak0
------------------------------------------------------
Both flights were of the same area, atl, spd, etc, only difference being the 1st was with a PowerShot N and the 2nd was with a SX260. 

I used the time offset geotagging function in mission planner to associate the gps data to the images. I had mismatched image to CAM message files so I could not use that function. 

For some reason my pixhawk did not record dataflshlogs for the two mission. So I had to use the tlogs for the gps data.

--------------------------------------------------
Here's the general flight information: 

Alt: 40m
Speed: 5m
Photo every: 4 sec
Distance between images: 20m
Area: 21082 square meters
--------------------------------------------------
Results after processing with ODM: 

**SX260:** 

This dataset got through all the processing steps in ODM with an error, but the results were sad :( see google drive...

 **PowerShot N:** 

I got an error not long into the processing. Here's where things went sideways...

[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[WARNING] Initial residual too low: 0 < 0.000001
[pcl::io::savePLYFile] Input point cloud has no data!
Error in OdmMeshing:
Error when saving mesh to file:
/home/aquila/Desktop/powershot1_geo/odm_meshing/odm_mesh.ply

For more detailed information, see log file.
[ERROR]   quitting cause: 
	/home/aquila/OpenDroneMap/build/bin/odm_meshing -inputFile /home/aquila/Desktop/powershot1_geo/pmvs/recon0/models/option-0000.ply -outputFile /home/aquila/Desktop/powershot1_geo/odm_meshing/odm_mesh.ply -logFile /home/aquila/Desktop/powershot1_geo/odm_meshing/odm_meshing_log.txt -maxVertexCount 100000 -octreeDepth 9 -samplesPerNode 1.0 -solverDivide 9
returned with code 256.

An error occurred. Check stdout above or the logs.

---------------------------------------------------

I could not get run.py to work so I ran the following while in the OpenDroneMap directory: 

sudo ./run.sh --project-path ~/Desktop/sx260_geo

----------------------------------------------------

The ODM I am using was a fresh install downloaded yesterday. 

I ran the caliterra dataset and got the normal results, but much quicker than I remember it taking the last time I used ODM, which was months ago. 

system specs: 

Ubuntu 14.04 64bit
RAM: 70.8GB
Intel X5680 @ 3.33ghz x 2
GPU: GTX960

---------------------------------------------------

I have no clue where my mistake is. Please help :( 

I tried to dig around to see if there were similar issues or results, It seemed like insufficient overlap might be the root cause of the poor results? But when I play back the flight in mission planner it looks like everything went well and there should be sufficient overlap. Like I said no clue :(

Thanks again for any assistance!

Fi156, 2016-11-06:
Overlap is an issue here: The screenshot from MissionPlanner states one possible problem already: Your image footprint size is somewhat around 55 by 40m. You planned the flight with in track distance of around 20m and between lines 22m. This is a barely 50% overlap... For the first datasets I would aim to get 75% overlap or even more...

aquilaimaging, 2016-11-10:
Thanks for the response. I redid the missions with 80% overlap and they came out quite good. Thanks again for the help!

dakotabenjamin, 2016-11-23:
@aquilaimaging As an aside I want to thank you for writing a concise and thorough detailing of your issue.

closed 2016-11-23 by dakotabenjamin