OpenDroneMap/ODM #419: Program fails: return code 256

jpmaniac, 2016-11-08:
I am running Ubuntu 14.04 in a Virtual machine, clean install on the dev branch and get a failure to complete

2016-11-08 09:41:57,397 reading features
2016-11-08 09:42:01,632 Merging features onto tracks
2016-11-08 09:42:01,632 Good tracks: 0
2016-11-08 09:42:01,944 0 partial reconstructions in total.
Traceback (most recent call last):
  File "/home/christian/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/home/christian/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
    reconstructions = data.load_reconstruction()
  File "/home/christian/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
    with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/home/christian/Drone/MyHouse80percOverlap/opensfm/reconstruction.json'
[ERROR]   quitting cause: 
	PYTHONPATH=/home/christian/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/christian/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/christian/Drone/MyHouse80percOverlap/opensfm
returned with code 256.

Any ideas on what that error code is? As you can see, ran it at 80% overlap, so there should be matches. I haven't even tried adding my GCPs yet....

pierotofy, 2016-11-11:
Would you be able to share the dataset? That way I could try to pin point the cause.

dakotabenjamin, 2016-12-06:
@jpmaniac Can you pull from master and try with a clean dataset? (not rerun)

closed 2018-03-28 by dakotabenjamin