OpenDroneMap/ODM #420: Import Error: libecto.so.0.6: cannot open shared object file, no such file or directory

CrYo-Dyne, 2016-11-09:
Hi Guys,

Sorry to bother with this issue, I have been trying to get my ODM to run on Ubuntu 14.04 Trusty 64-bit machine, placed my "images" folder inside OpenDroneMap folder, successful compilation and configuration. Once I run the command sudo python run.py --project-path /home/../.. where I point to my directory where images are located. 

I am then faced with the following output in the terminal:

Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/rick/OpenDroneMap/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

I have come across the same error on the "Issues" list on the forums here however with the types.py file for odm. I have copied the types.py's contents from the odmphoto_fix branch to my odm directory's types.py file and have tried to run from command line again but I am still faced with the same error.

Please can anyone advise on this I would really appreciate it :) And also thank you very much for this awesome project!

Best regards,
Rick

fitoprincipe, 2016-11-22:
I have the same problem. I made it work (half way unfortunately) using the command:
bash run.sh --project-path /home/... had other problems anyway, but may be you won't.
Good luck!

CrYo-Dyne, 2016-11-22:
Hi Fito,

Thanks so much for the response, I will give it a try as soon as I can and let you know what happens :)

Best regards,
Rick

dakotabenjamin, 2016-11-23:
Any luck?

luisgild, 2016-12-09:
Hi all.

Exactly the same problem here. I'm running ODM on a Ubuntu 16.04 LTS Vmware virtual machine (Windows 7 host).

dakotabenjamin, 2016-12-09:
This issue happens when the environment variables are not set properly. Make sure you follow the steps in the [installation](https://github.com/OpenDroneMap/OpenDroneMap#installation) to do that.

closed 2017-07-24 by dakotabenjamin