OpenDroneMap/ODM #421: import ecto When Logging on and off

GISkschultz, 2016-11-15:
I'm running 16.04 and after I install following the directions and run  python run.py --project-path /dronemap/ it works the first time around with no issues. I come into work the next day login into the rdp load it up to test it out again and I get this error 
python run.py -h
Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/dronemaster/DroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

One of the fixes I thought was changing types.py and changing line 30 as another issue had this didnt correct it.   

Going to the FAQ ./run.sh works and it also says make sure your envirofment variables are set correctly

are these two commands the enviroment variables?

export PYTHONPATH=$PYTHONPATH:`pwd`/SuperBuild/install/lib/python2.7/dist-packages:`pwd`/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:`pwd`/SuperBuild/install/lib

After I ran them it worked again however if I log off This issue will comeback, Can I automatically set this up so I dont have to continue running this command?

dakotabenjamin, 2016-11-15:
You can make it so that the environment variables are set every time you log in by adding those commands to ~/.bashrc. See http://unix.stackexchange.com/questions/117467/how-to-permanently-set-environmental-variables/117470#117470

GISkschultz, 2016-11-15:
Thank you for such a fast response!

closed 2016-11-23 by dakotabenjamin