OpenDroneMap/ODM #424: 'NoneType' object has no attribute 'replace' on calling model.replace

maxlein, 2016-11-20:
Hi,

i have a problem running the docker image on my mac.
Its the latest opendronemap image from git.

I call it like that:
`run -it --user root     -v /Users/max/Documents/visualSFM:/images     -v /Users/max/Documents/visualSFM/odm_orthophoto:/code/odm_orthophoto     -v /Users/max/Documents/visualSFM/odm_texturing:/code/odm_texturing     --rm opendronemap/opendronemap`

And this results in:

`[INFO]    Initializing OpenDroneMap app - Sun Nov 20 09:16:47  2016
[INFO]    Running ODM Load Dataset Cell
[DEBUG]   Loading dataset from: /images/images
Traceback (most recent call last):
  File "/code/run.py", line 38, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/dataset.py", line 57, in process
    self.params.force_ccd)
  File "/code/opendm/types.py", line 32, in __init__
    self.parse_pyexiv2_values(self.path_file, force_focal, force_ccd)
  File "/code/opendm/types.py", line 73, in parse_pyexiv2_values
    self.make_model = sensor_string(self.camera_make, self.camera_model)
  File "/code/SuperBuild/src/opensfm/opensfm/exif.py", line 63, in sensor_string
    model = model.replace(make, '') # remove possible duplicate 'make' information in 'model' for better matching
AttributeError: 'NoneType' object has no attribute 'replace'`

Somehow model is not set!?

dakotabenjamin, 2016-11-23:
I *think* this should be fixed in the master branch now, but I'm going to have to manually update the docker image (automated builds do not work anymore). Please follow the  [instructions here](https://github.com/OpenDroneMap/OpenDroneMap#build-and-run-using-docker) to build your own image.

dakotabenjamin, 2016-12-01:
Can you pull master, rebuild your image and test again?

closed 2018-03-28 by dakotabenjamin