OpenDroneMap/ODM #427: Issue with 0.6.16.zip & v21.2.zip

karigar, 2016-11-27:
-- Build files have been written to: /home/gsw/apps/odm/OpenDroneMap/SuperBuild/build
[  1%] [  2%] Performing download step (download, verify and extract) for 'catkin'
Performing download step (download, verify and extract) for 'gflags'
-- downloading...
     src='https://github.com/ros/catkin/archive/0.6.16.zip'
     dst='/home/gsw/apps/odm/OpenDroneMap/SuperBuild/download/0.6.16.zip'
     timeout='none'
CMake Error at /home/gsw/apps/odm/OpenDroneMap/SuperBuild/build/catkin/stamp/download-catkin.cmake:15 (file):
  file DOWNLOAD cannot open file for write.

-- downloading...
     src='https://github.com/gflags/gflags/archive/v2.1.2.zip'
     dst='/home/gsw/apps/odm/OpenDroneMap/SuperBuild/download/v2.1.2.zip'
CMakeFiles/catkin.dir/build.make:83: recipe for target 'catkin/stamp/catkin-download' failed
make[2]: *** [catkin/stamp/catkin-download] Error 1
     timeout='none'
CMakeFiles/Makefile2:60: recipe for target 'CMakeFiles/catkin.dir/all' failed
make[1]: *** [CMakeFiles/catkin.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
CMake Error at /home/gsw/apps/odm/OpenDroneMap/SuperBuild/build/gflags/stamp/download-gflags.cmake:27 (message):
  error: downloading 'https://github.com/gflags/gflags/archive/v2.1.2.zip'
  failed

    status_code: 1
    status_string: "Unsupported protocol"
    log: Protocol "https" not supported or disabled in libcurl

  Closing connection -1

  

CMakeFiles/gflags.dir/build.make:83: recipe for target 'gflags/stamp/gflags-download' failed
make[2]: *** [gflags/stamp/gflags-download] Error 1
CMakeFiles/Makefile2:200: recipe for target 'CMakeFiles/gflags.dir/all' failed
make[1]: *** [CMakeFiles/gflags.dir/all] Error 2
Makefile:75: recipe for target 'all' failed
make: *** [all] Error 2
Compiling build
CMake Warning at modules/odm_georef/CMakeLists.txt:16 (find_package):
  Could not find a package configuration file provided by "PCL" (requested
  version 1.7) with any of the following names:

    PCLConfig.cmake
    pcl-config.cmake

  Add the installation prefix of "PCL" to CMAKE_PREFIX_PATH or set "PCL_DIR"
  to a directory containing one of the above files.  If "PCL" provides a
  separate development package or SDK, be sure it has been installed.

CMake Error at modules/odm_georef/CMakeLists.txt:19 (find_package):
  Could not find a package configuration file provided by "OpenCV" with any
  of the following names:

    OpenCVConfig.cmake
    opencv-config.cmake

  Add the installation prefix of "OpenCV" to CMAKE_PREFIX_PATH or set
  "OpenCV_DIR" to a directory containing one of the above files.  If "OpenCV"
  provides a separate development package or SDK, be sure it has been
  installed.

-- Configuring incomplete, errors occurred!
See also "/home/gsw/apps/odm/OpenDroneMap/build/CMakeFiles/CMakeOutput.log".
Configuration Finished
gsw@gsw-Latitude-E4300:~/apps/odm/OpenDroneMap$

dakotabenjamin, 2016-11-28:
`log: Protocol "https" not supported or disabled in libcurl`
This seems to be the issue: https://curl.haxx.se/docs/faq.html#Protocol_xxx_not_supported_or_di

@smathermather should we default to http for package downloading, or expect users to use secure https?

smathermather, 2016-12-06:
I think https should be used if possible. I guess I don't quite understand the nature of the issue.

dakotabenjamin, 2016-12-06:
@karigar do you have other issues connecting through https protocol?

closed 2018-03-28 by dakotabenjamin