OpenDroneMap/ODM #429: IOError: [Errno 2] No such file or directory: './opensfm/reconstruction.json'

timvand, 2016-11-30:
While rerunning same job on a 16 CPU instance, received the same error. Previously ran the job on a 8 CPU instance without issue. Prior to running the job on the 16 CPU instance the only directory in place was the images dir.

What gives?

ERROR MESSAGE
2016-11-30 17:11:00,072 Matching DSC01982_geotag.JPG - 388 / 388
2016-11-30 17:11:00,363 reading features
2016-11-30 17:11:10,086 Merging features onto tracks
2016-11-30 17:11:10,086 Good tracks: 0
2016-11-30 17:11:10,367 0 partial reconstructions in total.
Traceback (most recent call last):
File "/root/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in 
command.run(args)
File "/root/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/mesh.py", line 22, in run
reconstructions = data.load_reconstruction()
File "/root/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 336, in load_reconstruction
with open(self.__reconstruction_file(filename)) as fin:
IOError: [Errno 2] No such file or directory: '/d01/projects/160613_BMM_M4-1608090019_odm/opensfm/reconstruction.json'
[ERROR] quitting cause:
PYTHONPATH=/root/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /root/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /d01/projects/160613_BMM_M4-1608090019_odm/opensfm
returned with code 256.

dakotabenjamin, 2016-11-30:
How much RAM is allocated for this machine?

timvand, 2016-11-30:
The 8 CPU instance has 32GB of ram and the 16 CPU has 64GB.

dakotabenjamin, 2016-11-30:
Assuming this is drone data, what overlap % did you fly at?

I think this is an OpenSfM issue and not an ODM issue, but I don't have enough information to give insight into the problem itself. You could try installing opensfm as its own instance and running it over the dataset (using the run_all command: https://github.com/mapillary/OpenSfM#running).

timvand, 2016-11-30:
All the imagery has been geo-tagged, initially processed as noted above, and completed all the processing of the 388 images in 4 hours time on an AWS m4x2 8 CPU/32GB Ubuntu 16.x instance.  This AM the instance was scaled up to an AWS m4x4 16 CPU/64GB Ubuntu 16.x instance and ran to the above noted error.

Have since re-cloned the binaries, ran configure.sh, run the job, and same error.

There were two other previous jobs which ran and completed, with lower image count, and geo-tagged.  No issues.

Will give the OpenSfM a run and see what happens.

dakotabenjamin, 2016-12-06:
Can you pull from master and try with a clean dataset? (not rerun)

timvand, 2016-12-06:
Sure.  I've since given up on the endeavor, and can easily re-constitute 
the setup on AWS with Ubuntu 16.

Reading through the bug list and fixes, are the original images, in the 
/images dir being modified?  If so, why?  And if that is the case, that 
means somewhere in the process, the original image set needs to be kept 
as a backup as well.

Regards,
Tim VanDermyden
888-831-4008

dakotabenjamin, 2016-12-06:
Now they should not be modified. I would always keep a backup of your images when working with experimental software anyways.

timvand, 2016-12-06:
Yeah, I do.  Always.

Was not aware the original geo-tagged images were being modified.

Regards,
Tim VanDermyden
888-831-4008

PR #433 (2016-12-06): Keeps original images pristine in georef
@mojodna fix for bug in #411 

Also should fix: #429, #426, #419
closed 2018-03-28 by dakotabenjamin