OpenDroneMap/ODM #431: Proposed Changes to Readme

pierotofy, 2016-11-30:
I was thinking of making a few changes to the Readme in order to help people get started with ODM more quickly. Part of these changes would include:

- Remove outdated references (for example, I don't think Vagrant is the best way to run ODM at this point, but the readme reports "see https://github.com/OpenDroneMap/odm_vagrant for running on Windows in a VM"
- Move the "Run OpenDroneMap", before the "Installation" section, merging parts of "Build and Run Using Docker" to be the default instructions for running ODM. I think this is important as I think many people skimming through the Readme would assume that the software requires Ubuntu. Now that we have nightly builds that are cross platform I think it would be easier to simply instruct people to get Docker and run `docker run -ti -v [...] opendronemap/opendronemap`.

Thought of asking for feedback before starting to write.

dakotabenjamin, 2016-11-30:
On one hand, vagrant tends to cause more issues than I'd like to handle. On the other hand, I think many people have a better time understanding a VM over Docker. On a third hand it's nice to keep everything in one repository (deleting odm_vagrant). 

On #430 I sort of did your second point by adding a "quickstart" instructions, but I didn't include docker instructions. There's no real "quickstart" to docker unless the user already has it installed. 

In all I'm only partially in favor of defaulting to Docker because of my own headaches trying to understand and use it. Maybe my experience is not universal though.

pierotofy, 2016-11-30:
The operating system makes a big difference in ease of installing docker. Linux is the most painful, Windows is so-so (sometimes works right away, sometimes requires a few tweaks), macOS is a breeze (I couldn't believe how easy it is to get docker up and running on there).

I would expect most Linux users to compile ODM from sources, unless they have Docker already installed. Perhaps the best way to organize this is to start the "Run OpenDroneMap" section with a "what OS are you using?" with anchor links to the proper section.

dakotabenjamin, 2016-11-30:
I didn't think of it that way, I only run docker from linux....

I think then that this is a good approach.

closed 2017-01-11 by pierotofy