OpenDroneMap/ODM #432: Blank spaces in orthophoto

Romi-Arg, 2016-12-01:
Dear all,
I'm quite new using ODM and I'm very exited with this project. 
I have run my first set of images (taken from a DJI Phantom 3 Pro) and compare the results with a previous orthomosaic I've done with other software using the very same images. 
At a first look the ODM's result seem fine but when I zoom in and compare with my previous orthomosaic I find some blank or white spaces. Some objects seem to be a little misplaced also. 

I leave here some captures for you to see: left is the orthomosaic I have from another software; right is the ODM orthophoto. 

[image]

[image]

[image]

I'm using ODM v.0.2 in Linux Mint and I processed the images using the Linux terminal (console). The only rare thing I found is that in many lines it said: "refinePatchBFGS Failed" . However, it continued processing. Don't know if it has something to do with this. 

[image]

Any ideas? 
Thank you!

pulquero, 2016-12-01:
Is the mesh sensible? Maybe you just need to tweak a few params. You could also try the dev branch, I've got much better results with it.

Romi-Arg, 2016-12-01:
The mesh seems to have the same problem. Where I see blank spaces in the orthophoto, in the mesh it has like "holes". Can you tell me about the dev branch?

pulquero, 2016-12-01:
You can just check it out from github and build it. It has all the latest improvements in it. The build is easy to do, but can take a bit of time.

Romi-Arg, 2016-12-01:
Ok. I'll try it then and see how it goes. Thanks!

Romi-Arg, 2016-12-01:
Actually, I compiled and installed my version of ODM following the steps described in the dev branch. Exactly these steps: https://github.com/OpenDroneMap/OpenDroneMap/tree/dev.

pulquero, 2016-12-01:
That is fine, but you need to switch branch first. BTW refinePatchBFGS error just means it had to switch solver. The dev branch uses a different solver altogether.

dakotabenjamin, 2016-12-01:
FYI, dev is depreciated and is currently *nearly* even with master. Use the master branch.

The problem here lies in the meshing algorithm. It is too simplistic for trees and buildings because IIRC it uses a poisson reconstruction. There are other, better algorithms that can handle trees (big round blobs that are up above the ground) and building (big angular blobs that extrude from the ground). What's keeping us currently from implementing these algorithms is that we don't have point classifications in place; the software doesn't know the difference between a tree point or a building point or a ground point. 

As such, all the buildings and trees look more like big balloon versions of themselves and when the texturing module tries to place the texture on top, it sort of tears the texture to fit.

Romi-Arg, 2016-12-01:
Thank you, @dakotabenjamin . So, to sum up: my problem here would be that the currently algorithm doesn't recognize very well all the different object's shapes (in this case trees and buildings). So, we'll have to wait for an improving in the algorithms and the software itself. Is that correct?

On the other hand, I'm a little bit confused now about the compilation and the steps to properly install the current version. Is it okey then to follow these steps and install version 0.2 (OpenDroneMap-v0_2-RC1.zip)? . This one (https://github.com/OpenDroneMap/OpenDroneMap) is the master branch, right?. 

I apologize for my "beginners" questions. I'm just trying to get familiar with GitHub and terminology.

smathermather, 2016-12-05:
@Romi-Arg -- ask many questions. They are welcome. If you are doing this with a DJI, you should pre-process the images per instructions by @dakotabenjamin, otherwise you will get texturing issues. My hunch is that once you pre-process the images, you'll see a much better result.

pulquero, 2016-12-05:
You could try lensfun (http://lensfun.sourceforge.net/) and something like darktable to preprocess the images. The latest version contains calibration data for a lot of common action cams include DJI drone cams.

smathermather, 2016-12-05:
Yup, lensfun might be another good option for preprocessing.

Romi-Arg, 2016-12-05:
Thank you all. I use Darktable to process images from my SLR Nikon cam. I never thought of doing this with the DJI's images, too. I will try this and also take a look to Lensfun.

dakotabenjamin, 2016-12-05:
My recommendation (right now) would be to simply delete your current directory, dowload OpenDroneMap-v0_2-RC1.zip, and install from there. If you want to improve the code or help us update ODM, then I would recommend cloning the repo with git (the master branch).

dakotabenjamin, 2016-12-05:
My preprocessing script only removes radial lens distortion. If your process does that, then [mine](https://github.com/dakotabenjamin/CameraCalibration) is largely irrelevant.

dakotabenjamin, 2018-03-28:
This issue is outdated and will be closed. Please open a new thread in http://community.opendronemap.org and reference this issue if you would like to continue the conversation.

closed 2018-03-28 by dakotabenjamin

ajitStephen, 2020-06-15:
@Romi-Arg did you got significant improvements done ? which values helped you ?

Romi-Arg, 2020-06-16:
Hi @ajitStephen!
I think the problem I had on that set of images was solved in the newest versions of ODM. I notice there is a great improvement comparing how ODM processed back in 2016 (when I opened the this issue) and how it works now.

smathermather, 2020-06-16:
The entire orthophoto process has been rebuilt in the last couple of years. :)

@ajitStephen -- feel free to reach out on https://community.opendronemap.org for questions!