OpenDroneMap/ODM #435: Use map-reduce to multi-thread georef::chooseBest*Triplet

pulquero, 2016-12-10:
These methods scale as N^3 so can be quite expensive. It should be possible to multi-thread them using map-reduce (map: transform each triple to error, reduce: take triplet with min error) or a multi-threaded version of std::transform/accumulate.

pulquero, 2016-12-11:
I'm considering trying to do a PR for this myself but it has been a long time since I have done any c++ (could do it in Java no prob). Any preference to what API I should use for threading/thread pools? If anyone knows openmp you could probably just add a couple of #pragmas and be done with it.

pierotofy, 2017-01-04:
Sorry for the late reply, at this point I would say use whatever API you are most comfortable with. We can always change it later (that's the beauty of OSS).

closed 2017-01-30 by pulquero