OpenDroneMap/ODM #437: error 32512

Moha555, 2016-12-14:
Hi...
I ran the ODM with different Data but every time I received an error as " returned with code 32512." and in project folder there wasn't any  *.ply or  *.jpeg file. 
Then I used sample data(odm_data_seneca).This time, after 

`[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
`

I received below error:

`sh: 1: /home/mhd/OpenDroneMap-master/SuperBuild/install/bin/texrecon: not found`
`[ERROR]   quitting cause:` 
`	/home/mhd/OpenDroneMap-master/SuperBuild/install/bin/texrecon `
`/home/mhd/Desktop/odm_data_seneca/pmvs/recon0/nvmCams.nvm `
`/home/mhd/Desktop/odm_data_seneca/odm_meshing/odm_mesh.ply `
`/home/mhd/Desktop/odm_data_seneca/odm_texturing/odm_textured_model -d gmi -o none     `
`returned with code 32512.`

`An error occurred. Check stdout above or the logs.  `

This time 
> odm-mesh.ply
 centers-all.ply
 centers-0000.ply

and 

 >option-000.ply

are created correctly but the folder of odm-texturing is empty. 
I just need a ortho-photo geo-referenced jpeg file.

What's the problem?
Is there an option that can creates it straightly without any additional processing ?

dakotabenjamin, 2016-12-14:
It looks like you ran into an installation error, ODM was unable to find the texrecon module. Make sure you have the most [up-to-date version of ODM](https://github.com/OpenDroneMap/OpenDroneMap/releases/tag/v0.2-RC1) and then run `bash configure.sh`.

Moha555, 2016-12-16:
@dakotabenjamin  Thank you for helping me ...
Last night, I used this source Code and ran same sample data on it.
at same time,I focused on outputs of terminal when I compiled it.
In first time, some syntax error of c++ code was shown but when I compiled it again, it did correctly without any error...!
Then, I ran ODM with that data and gave this error

>  Expanding patches... 
---- EXPANSION: 23 secs ---- 
Total pass fail0 fail1 refinepatch: 142904 74816 20714 47374 122190 
Total pass fail0 fail1 refinepatch: 100 52.354 14.495 33.1509 85.505 
FilterOutside 
mainbody: 
Gain (ave/var): 1.07534 0.530953 
2141465 -> 2141326 (99.9935%)	0 secs 
Filter Exact: ***************************************Killed 
[ERROR]   quitting cause: 
	/home/mhd/OpenDroneFolder/SuperBuild/install/bin/pmvs2 /home/mhd/Desktop/1/pmvs/recon0/ option-0000 
returned with code 35072. 
An error occurred. Check stdout above or the logs. 

and even ODM doesn't create odm_meshing and odm_texturing folders!!!!
What's your opinion of this problem?

Steps of my installation is based on [https://github.com/OpenDroneMap/OpenDroneMap/wiki/Installation](url)

Then I run this code in ODM directory : 

`python run.py --project-path mypath`
 
Is there a way to obtain ortho photo directly???

dakotabenjamin, 2016-12-19:
I'm glad your installation was a success. Can you tell me how much RAM you have on your machine? 

The orthophoto is the last step in this photogrammetric process and all the other steps are necessary to build up to that.

Moha555, 2016-12-20:
I used 8 GB RAM, but I can access to a pc with GB RAM.
Do you think It can be useful?

dakotabenjamin, 2016-12-20:
8 is generally fine. Try deleting the `pmvs` directory then running `python run.py --project path path --use-opensfm-pointcloud`

closed 2018-03-28 by dakotabenjamin