OpenDroneMap/ODM #438: GCP positioning problem

LucasMM86, 2016-12-18:
It seems that the georeferencing module doesn't find the correct position in the textured model for my GCPs.
My gcp-list.txt file looks like that:

WGS84 UTM 19S
500328.842 5452818.781 1000 921 690 IMG_3111_geotag.jpg      1
500328.842 5452818.781 1000 1842 690 IMG_3111_geotag.jpg      2
500328.842 5452818.781 1000 2763 690 IMG_3111_geotag.jpg      3
500328.842 5452818.781 1000 921 1380 IMG_3111_geotag.jpg      4
500328.842 5452818.781 1000 1842 1380 IMG_3111_geotag.jpg      5
500328.842 5452818.781 1000 2763 1380 IMG_3111_geotag.jpg      6
500328.842 5452818.781 1000 921 2070 IMG_3111_geotag.jpg      7
500328.842 5452818.781 1000 1842 2070 IMG_3111_geotag.jpg      8
500328.842 5452818.781 1000 2763 2070 IMG_3111_geotag.jpg      9

the image  'IMG_3111_geotag.jpg' looks like the attached image (I put the GCPs aproximately on the image, red points)
[image]

The position in model for gcps as reported by 'odm_georeferencing_log.txt' are in the other attached image.
[image]

I launched ODM with this command:
 	
python run.py --project-path /home/lucas/Documents/01_Aeromodelismo/01_Vant/03_Vuelos/20161112_C_Leones_vuelo2largo_5_GCP --rerun-from odm_georeferencing --odm_georeferencing-gcpFile /home/lucas/Documents/01_Aeromodelismo/01_Vant/03_Vuelos/20161112_C_Leones_vuelo2largo_5_GCP/gcp-list_9.txt --odm_georeferencing-useGcp --end-with odm_georeferencing

Does anywhere know about this problem? There is any solution or workaround?

dakotabenjamin, 2016-12-19:
Have you recently updated/pulled the latest changes? If not, I would recommend deleting your ODM install directory and downloading/installing the latest [release](https://github.com/OpenDroneMap/OpenDroneMap/releases/tag/v0.2-RC1) After that, check again the results and let us know how that changes. 

If you are using the latest release, is it possible to share the data? If privacy is necessary, we can share through other means.

LucasMM86, 2016-12-22:
Hello Benjamin and all,
I updated ODM and now it works fine. Thanks!
If you still want the images, they are here:
https://drive.google.com/drive/folders/0B7Xd1EabpUDGQzJHazhCS2hSZjQ?usp=sharing

I guess we should close this issue. I am new with GitHub so I dont know how to do that.

dakotabenjamin, 2016-12-22:
Great!

closed 2016-12-22 by dakotabenjamin