OpenDroneMap/ODM #439: How to apply some filters in a model generate by ODM?

andrespias, 2016-12-18:
Hello! I want to generate a vegetation-only map, applying some sort of filter in a DSM or dot cloud generated by ODM. I also want to get information about the altitude recorded at a specific point on the map. How can I do that?

dakotabenjamin, 2016-12-19:
Great question. That's not a current feature of ODM (Pull Requests encouraged!!) but it is certainly possible to do so using [PDAL](http://www.pdal.io/index.html). This library can both classify ground points and generate a DEM for getting elevation values.

andrespias, 2016-12-19:
Thanks @dakotabenjamin!! I've already trying to do something with PDAL with no results.
ODM, generates a point cloud as a ply file (pmvs/recon0/models/option-0000.ply) right?
When I try to run something like this I don't get any results:
docker run -v /srv/torreap:/data pdal/pdal:1.3 pdal ground /data/odm_data/caliterra/pmvs/recon0/models/option-0000.ply -o /data/ground.ply --filters.ground.classify=true
How can I convert the ODM output format into a allowed format by PDAL?
It's possible to get an output file in LAS format?

dakotabenjamin, 2016-12-19:
ODM will convert the point cloud to LAS but only if it is georeferenced. I believe the you can do this in PDAL by using the las writer.

smathermather, 2017-02-28:
@dakotabenjamin -- should we add DEM / DTMs to https://github.com/OpenDroneMap/OpenDroneMap/milestone/9 ?  I think they can be ready... .

smathermather, 2017-02-28:
For you, @dakotabenjamin: https://gist.github.com/smathermather/0f5f9d603f79d4a12fba70cb4be1f83c

closed 2017-04-24 by dakotabenjamin