OpenDroneMap/ODM #441: Error code 256. No module named csfm

luisgild, 2016-12-20:
Hi all...

I'm trying to run ODM on a virtual machine running Ubuntu 16.04. I get this error:

_from opensfm import features
File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/opensfm/features.py", line 12, in <module> import csfm
ImportError: No module named csfm
[ERROR]  quitting cause:
PYTHONPATH=/home/luis/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/luis/Proyectos-ODM/rozas/opensfm
returned with code 256._

Any help on this please?

dakotabenjamin, 2016-12-20:
Occasionally this module fails to install properly (I haven't isolated a reason). First try running 'configure.sh' again to see if that resolves the problem.

luisgild, 2016-12-20:
Thanks dakotabenjamin, I'll try.

luisgild, 2016-12-21:
Tried to run again 'configure.sh'. Now I  have a bit differente error:

_Traceback (most recent call last):
  File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 10, in <module>
    from opensfm import commands
  File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/__init__.py", line 2, in <module>
    import extract_metadata
  File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/extract_metadata.py", line 5, in <module>
    from opensfm import dataset
  File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 12, in <module>
    from opensfm import io
  File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/opensfm/io.py", line 9, in <module>
    from opensfm import features
  File "/home/luis/OpenDroneMap/SuperBuild/src/opensfm/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: No module named csfm
Traceback (most recent call last):
  File "run.py", line 50, in <module>
    plasm.execute(niter=1)
  File "/home/luis/OpenDroneMap/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/luis/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1_

dakotabenjamin, 2016-12-21:
Do you have opencv installed elsewhere on your machine?

closed 2020-04-14 by smathermather