OpenDroneMap/ODM #445: Do all large projects fail?

rmarkRCS, 2016-12-28:
I have tried several large datasets. They all fail like this (219 images):

(112,225)(55,0)(15,7)(8,320)(25,46)(82,0)(111,0)(31,9)(70,3)(90,1)(47,7)(23,0)(77,15)(117,0)(122,12)(126,0)(107,18)(44,29)(97,0)refinePatchBFGS failed!
(131,0)(19,150)(38,0)(1,52)(120,7)(4,0)(5,0)(121,0)(99,18)(21,100)(65,5)(50,13)(7,10)(98,2)refinePatchBFGS failed!
(20,1)(118,0)(95,54)(73,0)(10,1)(46,62)(100,4)(83,0)(113,6)(61,0)(48,1)(92,83)(64,53)(40,0)(124,0)(127,0)(63,0)(93,0)(87,42)(74,0)(78,0)(129,34)(34,0)(67,8)(36,0)(85,4)(6,0)(24,4)refinePatchBFGS failed!
(123,56)(125,0)(72,0)(54,77)(3,0)done
---- Initial: 0 secs ----
Total pass fail0 fail1 refinepatch: 1307098 37581 1022593 246924 284505
Total pass fail0 fail1 refinepatch: 100 2.87515 78.2338 18.891 21.7662
Expanding patches...Killed
Traceback (most recent call last):
  File "/code/run.py", line 50, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/pmvs.py", line 73, in process
    (context.pmvs2_path, tree.pmvs_rec_path))
  File "/code/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137

dakotabenjamin, 2016-12-28:
It is more likely that you do not have enough memory to run large project. PMVS is notorious for poor memory handling. Try `--use-opensfm-pointcloud` when running this data, and be patient as it will take a while.

rmarkRCS, 2016-12-28:
My Mac has 32 GB ram.
I will run it again, monitoring memory use.
I have modeled this data with PhotoScan.

What is "|use-opensfm-pointcloud"?

--Bob Mark
|

dakotabenjamin, 2016-12-28:
Perhaps memory is being allocated poorly to Docker (I assume you are using docker?). I'm not very well acquainted with docker use on a Mac. 

OpenSfM can perform the role of PMVS in the toolchain, but it is not turned on by default because it is a recent feature. It creates a dense point cloud by computing depth maps for each image. See: 

    Shen, S. “Accurate Multiple View 3D Reconstruction Using Patch-Based Stereo for Large-Scale Scenes.” IEEE Transactions on Image Processing 22, no. 5 (May 2013): 1901–14. doi:10.1109/TIP.2013.2237921.

sukoshi1507, 2016-12-28:
@rmarkRCS I'm interested in knowing how you are using ODM on a mac.

rmarkRCS, 2016-12-28:
Docker.

sukoshi1507, 2016-12-28:
Docker toolbox or Docker for Mac?
I'm asking since, i'm going to try the same soon : 

~ sukoshi$ git clone https://github.com/OpenDroneMap/OpenDroneMap.git
Cloning into 'OpenDroneMap'...
remote: Counting objects: 3747, done.
remote: Compressing objects: 100% (54/54), done.
cd OpenDroneMapts:  22% (853/3747), 260.01 KiB | 79.00 KiB/s   
Receiving objects:  22% (853/3747), 460.01 KiB | 82.00 KiB/s   
Receiving objects:  67% (2536/3747), 807.29 MiB | 122.00 KiB/s

rmarkRCS, 2016-12-28:
Docker for Mac. I had to increase its memory allocation; Went from 
default 2GB to 16GB, but may not need that much.

rmarkRCS, 2016-12-28:
Increased Docker memory allocation from default 2GB to 16GB. That worked, and was probably more than required.

dakotabenjamin, 2016-12-28:
Great? If you're issues are resolved, I will close this. Otherwise please reopen.

closed 2016-12-28 by dakotabenjamin

merkato, 2017-03-06:
```
(...) Merge depthmaps
Killed
Traceback (most recent call last):
  File "/home/mechanik/dev/odm/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/mechanik/dev/odm/OpenDroneMap/scripts/opensfm.py", line 90, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/mechanik/dev/odm/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 137
```
I'm trying to process 384 images. In first attempt OpenSfM stuck on cleaning depthmaps, with full swap file use. Then I've used --opensfm-processes 1. And results are as follow. 

```
Intel(R) Core(TM)2 Duo CPU     P8600  @ 2.40GHz
8 GB RAM, 8 GB swap file, Xubuntu 16.04
```
With same dataset in 200 images samples everything is ok.

dakotabenjamin, 2017-03-06:
@merkato that is the current fix as we work on backend memory issues.

bruizerbg, 2018-06-11:
Hi @dakotabenjamin - Your post on this topic is very useful - your  --use-opensfm-pointcloud looks like the solution I need but I am such a NEWBIE to ODM that I don't know where to use that handle.
I am using the WebODM client - where can I specify these options before running a job?
Thanks,

smathermather, 2018-06-11:
You can customize your options by choosing edit options, and scroll down to the appropriate option.

bruizerbg, 2018-06-11:
Hi @smathermather - I don't have an "edit options" choice in the WebODM client. Can you be more specific?
[image]

bruizerbg, 2018-06-11:
Thanks Stephen - I couldn't find any open that exactly said "--use-opensfm-pointcloud".
I'm trying one that appeared similar - unless it is labelled differently in the latest WebODM version.
Trying my image set now.
Thanks,
Bruce

smathermather, 2018-06-11:
👍

gpetty, 2018-10-16:
I'm getting the same error running Docker/ODP (for the first time) from the command line on a Mac.  I have tried increasing the memory allocation for Docker from 2GB to 8GB, but that didn't solve the problem.  It's not clear to me whether that allocation is for physical memory or virtual memory; my Mac has only 16GB of RAM, so I didn't want to push it.

Above, the advice is also given to use --use-opensfm-pointcloud , but that option doesn't seem to be recognized in my command line, at least in the positions where I have tried placing it.  For reference,  I'm using Docker Community Edition Version 18.06.1-ce-mac73 (26764).   The original command line I'm using is the same as that given on the Wiki except for the image path:

docker run -it --rm -v  "/Users/gpetty/Dropbox/Flying/UAS/FLIR/20181015_162846:/code/images" -v "$(pwd)/odm_orthophoto:/code/odm_orthophoto" -v "$(pwd)/odm_georeferencing:/code/odm_georeferencing" opendronemap/opendronemap --mesh-size 100000 

The number of images is large (not quite 1200), but the images are relatively small (640x512), which I hoped might mean that the memory requirement would be less than for higher-resolution images.

The last lines displayed before the error message are:
...
Starting 1009/1010 ID: 1183 Neighbors: 887 717 479 921 242 570 
Finished 1006/1010 ID: 1180
Starting 1010/1010 ID: 1184 Neighbors: 61 946 498 138 1026 167 
Finished 1007/1010 ID: 1181
Finished 1009/1010 ID: 1183
Finished 1010/1010 ID: 1184
Saving views to MVE files... done.
Done. Reconstruction took: 24254.6s
Generating Pointcloud, Cutting surfaces for 1020 views ...
Killed

smathermather, 2018-10-17:
That many images, even small ones, would definitely overwhelm 8GB of RAM. You can try using the flag ```--opensfm-processes 1``` to reduce the number of threads. It might help.

Alternatively (if the above fails), look to the split-merge approach: http://docs.opendronemap.org/large.html

Also, this is much better conversation to have on http://community.opendronemap.org rather than here on a closed issue. :)

Follow up there and let us know how it went!

smathermather, 2018-10-22:
@jpwhitney -- can you please start a thread on community.opendronemap.org? That is a far better place to have this conversation.  Thanks!