OpenDroneMap/ODM #448: Bug in configure.sh? on

triceam, 2017-01-03:
I'm not sure if this is a bug, or a configuration issue on my end...  I am updating an Ubuntu 14.04 server to the latest on master (last update on the server was about 4 months ago).  However, when I run configure.sh, I am seeing this message showing pcl fails to compile:

```
Installing OpenDroneMap Dependencies
Compiling SuperBuild
-- SuperBuild files will be downloaded to: /root/OpenDroneMap/SuperBuild/download
-- SuperBuild source files will be extracted to: /root/OpenDroneMap/SuperBuild/src
-- SuperBuild source files will be installed to: /root/OpenDroneMap/SuperBuild/install
-- SuperBuild binary files will be located to: /root/OpenDroneMap/SuperBuild/build
-- OpenCV 2.4.11 not found --- Adding internal version
-- PCL 1.7.2 found
-- GFlags 2.1.2 not found --- Adding internal version
-- Ceres 1.10.0 not found --- Adding internal version
-- OpenGV force build --- Adding internal version
-- OpenSfM force build --- Adding internal version
-- CMVS force build --- Adding internal version
-- Catkin force build --- Adding internal version
-- Ecto force build --- Adding internal version
-- PDAL force build --- Adding internal version
-- MvsTexturing force build --- Adding internal version
-- Configuring done
-- Generating done
-- Build files have been written to: /root/OpenDroneMap/SuperBuild/build
[ 10%] [ 20%] Built target catkin
Built target gflags
[ 30%] [ 40%] Built target opencv
Built target cmvs
[ 50%] Built target mvstexturing
[ 70%] [ 70%] Built target opengv
Built target pdal
[ 80%] [ 90%] Built target ecto
Built target ceres
[100%] Built target opensfm
Compiling build
-- Could NOT find Boost
CMake Warning at SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:31 (message):
  common is required but boost was not found
Call Stack (most recent call first):
  SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:482 (pcl_report_not_found)
  SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:623 (find_external_library)
  modules/odm_georef/CMakeLists.txt:16 (find_package)

-- Could NOT find Boost
CMake Warning at SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:31 (message):
  common is required but boost was not found
Call Stack (most recent call first):
  SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:482 (pcl_report_not_found)
  SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:623 (find_external_library)
  modules/odm_meshing/CMakeLists.txt:11 (find_package)

-- Could NOT find Boost
CMake Error at SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:29 (message):
  common is required but boost was not found
Call Stack (most recent call first):
  SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:482 (pcl_report_not_found)
  SuperBuild/install/share/pcl-1.7/PCLConfig.cmake:623 (find_external_library)
  modules/odm_orthophoto/CMakeLists.txt:12 (find_package)
```

I've tried updating all the latest dependencies, and have verified that I have boost 1.55 installed on the server.  

I have tried pulling the latest release snapshot off of github, and the latest on `master`, but no luck.   Possibly a dependency mismatch somewhere?   It's not immediately obvious to me where.

pierotofy, 2017-01-04:
Perhaps boost is installed in a non-standard location? Is there a /usr/include/boost/ folder on your system?

triceam, 2017-01-05:
Yes, `/usr/include/boost` does exist.   Anyone familiar with boost know how to determine which version, and how i can see if it is the right version for ODM?

pierotofy, 2017-01-05:
Maybe an obvious question, but are you starting the build process from a clean build directory?

triceam, 2017-01-05:
Have tried a few variations:

- new, clean directory
- update existing directory (as-is)
- update existing directory (removed build folder)

pierotofy, 2017-01-05:
I think it's the newer version of Boost that is not supported by the PCL dependency pulled in by Superbuild.

Try to manually edit the line in SuperBuild/install/share/pcl-1.7/PCLConfig.Cmake:

```
  if(${CMAKE_VERSION} VERSION_LESS 2.8.5)
    SET(Boost_ADDITIONAL_VERSIONS "1.43" "1.43.0" "1.44" "1.44.0" "1.45" "1.45.0" "1.46.1" "1.46.0" "1.46" "1.47" "1.47.0")
```

To include your version of Boost:

```
  if(${CMAKE_VERSION} VERSION_LESS 2.8.5)
    SET(Boost_ADDITIONAL_VERSIONS "1.43" "1.43.0" "1.44" "1.44.0" "1.45" "1.45.0" "1.46.1" "1.46.0" "1.46" "1.47" "1.47.0", "1.55.0")
```

If other parts of the compilation fail, you might want to consider downgrading your Boost library to 1.47.

triceam, 2017-01-06:
No luck.  Tried that, tried removing every trace of boost and reinstalling.  `sudo apt-get install libboost-dev-all` installs both 1.54 and 1.55.  Both added to `PCLConfig.make` and same error compiling `pcl-1.7`.  :/

pierotofy, 2017-01-06:
1.54 and 1.55 are not supported by PCL. You might have to manually install boost from sources, or install it from a previous Ubuntu release.

triceam, 2017-01-10:
This is still perplexing.  I have Boost 1.47 installed, but still exact same error:

```
$ cat /usr/include/boost/version.hpp | grep "BOOST_LIB_VERSION"
//  BOOST_LIB_VERSION must be defined to be the same as BOOST_VERSION
#define BOOST_LIB_VERSION "1_47"
```

pierotofy, 2017-01-10:
Mm, I'm out of ideas. A workaround would be to use docker, but I'm not sure if it's ideal for your case.

triceam, 2017-01-10:
Yeah, this thing is a beast of a server, and I want to leverage every bit of computing power it has to offer without additional abstraction layers.  I thought about docker, but i think that's a last resort.

triceam, 2017-01-11:
I ended up reinstalling the OS, and ODM installed just fine.  I really don't know what happened, but there was definitely a configuration issue somewhere on my end.

closed 2017-01-11 by triceam

juvinski, 2017-03-01:
Hi,

I found the same problem, edit the CMakeLists.txt inside SuperBuild folder and change all flags
option(ODM_BUILD_XXXXXXX "Force to build OpenCV library" OFF) to ON

option(ODM_BUILD_OpenCV "Force to build OpenCV library" ON)