OpenDroneMap/ODM #449: ImportError: libecto.so.0.6

wheelie56, 2017-01-08:
Hello All,

Does anyone have some insight into this error condition? I just upgraded to Ubuntu 16.04 and then installed and configured OpenDroneMap with no errors.  I then attempted to run a test project and get the Python import error seen below. Thank you so much.
_____________________________________________________________________________________
export PYTHONPATH=$PYTHONPATH:/home/maddog/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
export PYTHONPATH=$PYTHONPATH:/home/maddog/OpenDroneMap/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/home/maddog/OpenDroneMap/SuperBuild/install/lib
____________________________________________________________________________________
maddog@maddog-1525:~/OpenDroneMap$ python run.py --project-path /home/maddog/imagery/odm_data_caliterra/images
____________________________________________________________________________________
Traceback (most recent call last):
  File "run.py", line 8, in <module>

import ecto
  File "/home/maddog/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>

from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup

ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

wheelie56, 2017-01-08:
Ok. I had to change 

PYTHONPATH=$PYTHONPATH:/home/maddog/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages

to 

PYTHONPATH=$PYTHONPATH:/usr/lib/python2.7/dist-packages

to get the ball rolling.  So my question now is... are there others environment variables to change be cause of the change above?  After I started processing it stopped as seen below. 

[INFO]    Resized 77 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   running PYTHONPATH=/home/maddog/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/maddog/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/maddog/imagery/odm_caliterra/opensfm
Traceback (most recent call last):
  File "/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 10, in <module>
    from opensfm import commands
  File "/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/__init__.py", line 2, in <module>
    import extract_metadata
  File "/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/extract_metadata.py", line 5, in <module>
    from opensfm import dataset
  File "/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dataset.py", line 12, in <module>
    from opensfm import io
  File "/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/io.py", line 9, in <module>
    from opensfm import features
  File "/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/features.py", line 12, in <module>
    import csfm
ImportError: No module named csfm
[ERROR]   quitting cause: 
	PYTHONPATH=/home/maddog/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/maddog/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/maddog/imagery/odm_caliterra/opensfm
returned with code 256.

wheelie56, 2017-01-08:
Leave a comment

dakotabenjamin, 2017-01-11:
First try rerunning `configure.sh` to see if any errors pop up. 

Please type 
```bash
echo $PYTHONPATH
echo $LD_LIBRARY_PATH
```
and paste the results here.

wheelie56, 2017-01-12:
Re-ran configure.sh and the opensfm does not build.  The results of the echo returned nothing.  I also did a printenv and I do not see that these environment variables are set.  They are in my bashrc file as per your instructions.

dakotabenjamin, 2017-01-12:
You have to log out and back in before the environment variables are set. Also make sure you have downloaded the most recent version of ODM: https://github.com/OpenDroneMap/OpenDroneMap/releases/download/v0.2/OpenDroneMap-v0_2.zip

Regarding opensfm, I need to know the specific errors being thrown. When you run configure.sh, can you paste the output that shows where it fails?

wheelie56, 2017-01-13:
/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/depthmap.cc:386:9: error: ‘class std::vector<cv::Vec<double, 3> >’ has no member named ‘emplace_back’
     ts_.emplace_back(pt);
         ^
/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/depthmap.cc:387:16: error: ‘class std::vector<std::vector<int> >’ has no member named ‘emplace_back’
     neighbors_.emplace_back(neighbors);
                ^
/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/depthmap.cc:388:13: error: ‘class std::vector<cv::Mat>’ has no member named ‘emplace_back’
     depths_.emplace_back(cv::Mat(height, width, CV_32F, (void *)pdepth).clone()
             ^
/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/depthmap.cc:389:13: error: ‘class std::vector<cv::Mat>’ has no member named ‘emplace_back’
     planes_.emplace_back(cv::Mat(height, width, CV_32FC3, (void *)pplane).clone
             ^
/home/maddog/OpenDroneMap/SuperBuild/src/opensfm/opensfm/src/depthmap.cc:390:13: error: ‘class std::vector<cv::Mat>’ has no member named ‘emplace_back’
     colors_.emplace_back(cv::Mat(height, width, CV_8UC3, (void *)pcolor).clone(

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
There are lots of errors from here up, I included a small sample size above...
--------------------------------------------------------------------------

CMakeFiles/csfm.dir/build.make:62: recipe for target 'CMakeFiles/csfm.dir/csfm.cc.o' failed
make[5]: *** [CMakeFiles/csfm.dir/csfm.cc.o] Error 1
CMakeFiles/Makefile2:142: recipe for target 'CMakeFiles/csfm.dir/all' failed
make[4]: *** [CMakeFiles/csfm.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make[3]: *** [all] Error 2
CMakeFiles/opensfm.dir/build.make:116: recipe for target 'opensfm/stamp/opensfm-build' failed
make[2]: *** [opensfm/stamp/opensfm-build] Error 2
CMakeFiles/Makefile2:365: recipe for target 'CMakeFiles/opensfm.dir/all' failed
make[1]: *** [CMakeFiles/opensfm.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2

pierotofy, 2019-05-23:
These issues (especially for ecto) should be now working. If not, please open a new issue! :hugs:

closed 2019-05-23 by pierotofy