OpenDroneMap/ODM #450: ValueError running ODM in Docker on Mac

shenuhcide, 2017-01-08:
I downloaded and installed docker and was able to successfully run ODM on the test data included.

However, when I ran my own image data using this command:

`docker run -it --user root -v /Users/shenuhcide/Desktop/OpenDroneMap/UAV_Imagery:/code/images -v /Users/shenuhcide/Desktop/OpenDroneMap/odm_orthophoto_uav:/code/odm_orthophoto -v /Users/shenuhcide/Desktop/OpenDroneMap/odm_texturing_uav:/code/odm_texturing --rm odm_image`

And it wrote the texturing files, but then I got this error before the orthophoto images were written:

`Traceback (most recent call last):
  File "/code/run.py", line 50, in <module>
    plasm.execute(niter=1)
  File "/code/scripts/odm_georeferencing.py", line 122, in process
    geo_ref.parse_coordinate_system(tree.odm_georeferencing_coords)
  File "/code/opendm/types.py", line 304, in parse_coordinate_system
    self.utm_zone = int(ref[2][:len(ref) - 1])
ValueError: invalid literal for int() with base 10: '5N'`

Any assistance would be greatly appreciated.

Thanks!

jrleeman, 2017-01-09:
Are you using a GCP file? If so what does it look like?

shenuhcide, 2017-01-09:
I am not using a gcp file.

Sheina B. Sim, Ph.D.
Junior Faculty
University of Hawaii at Manoa and USDA-ARS Daniel K. Inouye US PBARC
Phone: 808-932-2175

jrleeman, 2017-01-10:
Hmm. I'm not sure then! Could be something in the EXIF data - are you at least getting a textured mesh out before it dies? I'm new to this package myself, so someone else may have to step in.

shenuhcide, 2017-01-10:
Yes, I'm getting textured output files.

Sheina B. Sim, Ph.D.
Junior Faculty
University of Hawaii at Manoa and USDA-ARS Daniel K. Inouye US PBARC
Phone: 808-932-2175

dakotabenjamin, 2017-01-11:
Are you able to share your dataset so I can try to reproduce the error? 

Otherwise can you run exiftool on a sample image and paste the output? eg.:
```bash
exiftool -a -gps:all <image.jpg>
```

jleichty, 2017-05-18:
I experienced a similar error today (sample dataset ran fine, but my own did not). See below for the relevant output.

```
[INFO]    Running ODM Texturing Cell - Finished
[INFO]    Running ODM Georeferencing Cell
[DEBUG]   None
[WARNING] No coordinates file. Generating coordinates file: /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_georeferencing/coords.txt
[INFO]    Running georeferencing with generated coords file.
[DEBUG]   running /home/jleichty/sandbox/OpenDroneMap_v0_3_1/build/bin/odm_georef -bundleFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/opensfm/bundle_r000.out -inputCoordFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_georeferencing/coords.txt -inputFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_texturing/odm_textured_model.obj -outputFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_texturing/odm_textured_model_geo.obj -inputPointCloudFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/opensfm/depthmaps/merged.ply -outputPointCloudFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_georeferencing/odm_georeferenced_model.ply  -logFile /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_georeferencing/odm_georeferencing_log.txt -georefFileOutputPath /home/jleichty/sandbox/odm_projects/test_site_2017-04-30/odm_georeferencing/odm_georeferencing_model_geo.txt
[DEBUG]   Line: WGS84 UTM 4N

Traceback (most recent call last):
  File "/home/jleichty/sandbox/OpenDroneMap_v0_3_1/run.py", line 46, in <module>
    plasm.execute(niter=1)
  File "/home/jleichty/sandbox/OpenDroneMap_v0_3_1/scripts/odm_georeferencing.py", line 147, in process
    geo_ref.parse_coordinate_system(tree.odm_georeferencing_coords)
  File "/home/jleichty/sandbox/OpenDroneMap_v0_3_1/opendm/types.py", line 339, in parse_coordinate_system
    self.utm_zone = int(ref[2][:len(ref) - 1])
ValueError: invalid literal for int() with base 10: '4N'
```

Here is the exiftool output for a sample image:
```
jleichty@vu:~/sandbox/odm_projects/test_site_2017-04-30/images$ exiftool -a -gps:all GOPR1224.JPG 
GPS Version ID                  : 2.3.0.0
GPS Latitude Ref                : North
GPS Latitude                    : 71 deg 23' 47.24"
GPS Longitude Ref               : West
GPS Longitude                   : 156 deg 25' 9.54"
GPS Altitude Ref                : Above Sea Level
GPS Altitude                    : 44.610001 m
```

It's worth noting that I "manually" added GPS data to these images (taken by a GoPro camera that doesn't have GPS) with exiftool.

dakotabenjamin, 2018-03-28:
I think this is fixed, please reopen if not.

closed 2018-03-28 by dakotabenjamin