OpenDroneMap/ODM #453: Out of memory errors with use-opensfm-pointcloud

triceam, 2017-01-12:
I'm getting out of memory errors on a machine with 64 gigs of RAM using the langley dataset with the opensfm pointcloud option

Invoked like so:
`./run.sh --project-path ../odm_data/langley/ --use-opensfm-pointcloud`

Everything seems fine, then I see this 
```
2017-01-12 10:39:59,362 Using precomputed clean depthmap IMG_0686.jpg
2017-01-12 10:39:59,362 Using precomputed clean depthmap IMG_0629.jpg
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 326, in _handle_workers
    pool._maintain_pool()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 230, in _maintain_pool
    self._repopulate_pool()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 223, in _repopulate_pool
    w.start()
  File "/usr/lib/python2.7/multiprocessing/process.py", line 130, in start
    self._popen = Popen(self)
  File "/usr/lib/python2.7/multiprocessing/forking.py", line 121, in __init__
    self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory

2017-01-12 10:40:06,047 Using precomputed clean depthmap IMG_0696.jpg
2017-01-12 10:40:06,047 Using precomputed clean depthmap IMG_0719.jpg
2017-01-12 10:40:12,935 Using precomputed clean depthmap IMG_0527.jpg
2017-01-12 10:40:12,935 Using precomputed clean depthmap IMG_0617.jpg
2017-01-12 10:40:19,222 Using precomputed clean depthmap IMG_0535.jpg
2017-01-12 10:40:19,226 Using precomputed clean depthmap IMG_0548.jpg
2017-01-12 10:40:25,876 Using precomputed clean depthmap IMG_0783.jpg
2017-01-12 10:40:25,882 Using precomputed clean depthmap IMG_0569.jpg
Traceback (most recent call last):
  File "/root/OpenDroneMap/SuperBuild/src/opensfm/bin/opensfm", line 34, in <module>
    command.run(args)
  File "/root/OpenDroneMap/SuperBuild/src/opensfm/opensfm/commands/compute_depthmaps.py", line 25, in run
    dense.compute_depthmaps(data, graph, reconstructions[0])
  File "/root/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 38, in compute_depthmaps
    parallel_run(clean_depthmap, arguments, processes)
  File "/root/OpenDroneMap/SuperBuild/src/opensfm/opensfm/dense.py", line 50, in parallel_run
    return p.map(function, arguments)
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 251, in map
    return self.map_async(func, iterable, chunksize).get()
  File "/usr/lib/python2.7/multiprocessing/pool.py", line 558, in get
    raise self._value
MemoryError: out of memory
2017-01-12 10:40:36,023 Using precomputed clean depthmap IMG_0713.jpg
2017-01-12 10:40:36,024 Using precomputed clean depthmap IMG_0748.jpg
Traceback (most recent call last):
  File "/root/OpenDroneMap/run.py", line 50, in <module>
    plasm.execute(niter=1)
  File "/root/OpenDroneMap/scripts/opensfm.py", line 90, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/root/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 1
```

triceam, 2017-01-12:
This is using the latest snapshot from `master`

dakotabenjamin, 2017-01-12:
How many CPUs? see #452

triceam, 2017-01-12:
OK, good to know about the cpu limit2 CPUs, each has 12 cores, each core has 2 threads, so 48 threads of execution.

Any guess at how many threads is a reasonable amount if using `--opensfm-processes`?

dakotabenjamin, 2017-01-12:
Since it was just added, I'm still exploring that.

dakotabenjamin, 2017-01-12:
I had trouble with 12 cores on a 12Gb VM so I cut to 6

triceam, 2017-01-12:
OK thx. I'm starting at 24, we'll see how it goes.  If i run out of memory again, will drop in half again.

triceam, 2017-01-12:
Still running `run.sh`, but it looks like it completed the OpenSFM part of the process.  `--opensfm-processes` took about 42GB RAM

closed 2017-01-12 by triceam