OpenDroneMap/ODM #455: Memory Leak during meshing

dakotabenjamin, 2017-01-18:
I want to report a possible memory leak during the meshing step which causes failure at the next step. See below for screnshot of the memory of the machine and the run log. 

[image]

```bash
[INFO]    Running ODM Meshing Cell - Finished
[INFO]    Running MVS Texturing Cell
[DEBUG]   Writing MVS Textured file in: /home/dmb2/sukoshi-uas/odm_texturing/odm_textured_model.obj
[DEBUG]   running /home/dmb2/opendronemap/SuperBuild/install/bin/texrecon /home/dmb2/sukoshi-uas/opensfm/reconstruction.nvm /home/dmb2/sukoshi-uas/odm_meshing/odm_mesh.ply /home/dmb2/sukoshi-uas/odm_texturing/odm_textured_model -d gmi -o none     
/home/dmb2/opendronemap/SuperBuild/install/bin/texrecon (built on Jan 11 2017, 13:18:53)
Load and prepare mesh: 
PLY Loader: comment PCL generated
Reading PLY: 81777 verts... 163344 faces... done.
Warning: Zero-length normals detected: 34 face normals, 1 vertex normals
Generating texture views: 
NVM: Loading file...
NVM: Number of views: 599
NVM: Number of features: 0
        Loading 2%...Received signal SIGSEGV (segmentation fault)               
Segmentation fault (core dumped)
Traceback (most recent call last):
  File "/home/dmb2/opendronemap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/dmb2/opendronemap/scripts/mvstex.py", line 106, in process
    '{keepUnseenFaces}'.format(**kwargs))
  File "/home/dmb2/opendronemap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 139
dmb2@opendronemap-gui:~/opendronemap$ ./run.sh --project-path ~/sukoshi-uas/ --use-opensfm-pointcloud --opensfm-processes 6 --rerun-from mvs_texturing

```

Rerunning from mvs_texturing was successful. My guess is that there is a memory leak from meshing that led to a lack of memory available for mvs_texturing to run.

xialang2012, 2017-01-19:
The memory will be released when the process end. Does it the memory management strategy of the system.

dakotabenjamin, 2017-01-20:
Do you think there is a way to better manage the memory so it doesn't do this?

closed 2020-04-14 by smathermather