OpenDroneMap/ODM #456: Ideal hardware for running ODM?

snowsecure, 2017-01-23:
I'm curious what the ideal hardware is for running ODM. More cores, faster cores, better GPU, more RAM, etc?? What's your wish list?

I'm thinking of pairing a 6950x with 64GB of RAM to chew through my data sets. Am I heading in the right direction, or would I be better served with older, slower, lower core count dual Xeons?

Thanks.

dakotabenjamin, 2017-01-23:
Something to consider is that you can run into issues with virtual cpus (like with the xeons) that will actually hamper performance during matching/reconstruction. 

In some other cases, the more cores you have, the greater the memory usage. The 6950x and 64GB RAM is probably a good bet. I run ODM on either a Xeon E5-1650 (3.50GHz) with 32GB RAM or a server with 16 vCPU Xeon E5-2670 (2,50GHz) and 44GB RAM. Very large datasets I will run on the server because it has more RAM but it tends to work slightly slower. 

We do not have GPU processing yet, and no plans to add that feature in the near future.

dakotabenjamin, 2017-04-03:
Closing, see #516

closed 2017-04-03 by dakotabenjamin