OpenDroneMap/ODM #457: equirectengular panoramas

polto, 2017-01-24:
Hi !
I am trying to run ODM on equirectengular full sphere panoramic images.
It seems openSfM support equirectengular panoramas, but I can not run ODM in a way it runs openSfM with "projection_type": "equirectengular" in opensfm/camera_models.json 
Anybody have any experience on using ODM on panoramas ?
thanks !

dakotabenjamin, 2017-01-24:
What errors are you getting?

polto, 2017-01-24:
no errors, but the reconstructions are nonsense.
and I can see that openSfM supports EQR images - https://github.com/mapillary/OpenSfM/issues/77

dakotabenjamin, 2017-01-24:
Can you upload the reconstruction and/or the images? I would be interested in seeing how this works.

polto, 2017-01-25:
Sorry. It took some time.
Here is a data sample - https://github.com/3d4pi/3d4pi-data-sample1/

dakotabenjamin, 2017-01-26:
I think that this is a use case that ODM is not made to handle. 

The issue is not really that it isn't possible to reconstruct. Technically this is true, ODM will run from start to finish and something will result. In this case, however, the input data has some issues that ODM does not account for. ODM does not mask out unwanted portions of the images, and will attempt to reconstruct from the whole thing. For instance, having any sky or horizon will generally ruin a reconstruction. For that reason, 360 cameras are not ideal for this type of software. 

That being said, I was able to follow the advice from the mapillary thread to reconstruct a sparse point cloud (unable to go further because these images don't have a focal length in EXIF, which is required), but I'm having trouble viewing it. You can follow the instructions for viewing the sparse point cloud [here](https://geokota.wordpress.com/2016/06/23/viewing-sparse-point-clouds-from-opendronemap/).

pierotofy, 2020-02-03:
Try to process with `--camera-lens spherical`. It should process correctly (this has been added a while ago)! If not, please re-open the issue. :pray:

closed 2020-02-03 by pierotofy