OpenDroneMap/ODM #458: Can't run ODM - libecto.so.0.6 error

Romi-Arg, 2017-01-24:
Hi everyone!
I'm having some trouble running ODM. I've installed OpenDroneMap-v0_2 following all the instructions and it seems to be okey. No error messages.

However, when I tried to run ODM using "python run.py --project-path /path/to/project -i /path/to/images", I get this error:

Traceback (most recent call last):
  File "run.py", line 8, in <module>
    import ecto
  File "/home/romina/OpenDroneMap-v0_2/SuperBuild/install/lib/python2.7/dist-packages/ecto/__init__.py", line 30, in <module>
    from ecto.ecto_main import _cell_base, _cell_cpp, __getitem_list__, __getitem_slice__, __getitem_str__, __getitem_tuple__, lookup
ImportError: libecto.so.0.6: cannot open shared object file: No such file or directory

Am I doing something wrong? I used a few weeks ago ODM and I had no problems. 
Thank you all!

dakotabenjamin, 2017-01-24:
Try this instead: "./run.sh --project-path /path/to/project"

Romi-Arg, 2017-01-24:
Thank you for your answer, @dakotabenjamin.
However now using that command gives me the "permission denied" error, even though I use "sudo" as superuser. 
Any ideas?

dakotabenjamin, 2017-01-24:
Can you be more specific about the error by copy-pasting the output?

When you ran weeks ago, did you set your environment variables in ~/.bashrc? Are they still there? Are you able to manually set the environment variables?
```
export PYTHONPATH=$PYTHONPATH:/your/path/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages
export PYTHONPATH=$PYTHONPATH:/your/path/OpenDroneMap/SuperBuild/src/opensfm
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/your/path/OpenDroneMap/SuperBuild/install/lib
```

Romi-Arg, 2017-01-24:
May be this helps: 

[image]

Trying both ways, I get different errors.

When I used ODM the first time, I've set the environment variables. After I read your answer, I checked and somehow they were gone. Don't know why. Now I set them manually like this:
[image]

dakotabenjamin, 2017-01-24:
OK great. Run `source ~/.bashrc` then you can try using the python script again.

Romi-Arg, 2017-01-24:
Great! Thank you a lot, @dakotabenjamin !

closed 2017-01-24 by Romi-Arg