OpenDroneMap/ODM #460: Run Exception: opendm/system.py line 28 child returned {}

bryanibit, 2017-01-26:
[INFO]    Resized 57 images
[INFO]    Running ODM Resize Cell - Finished
[INFO]    Running ODM OpenSfM Cell
[DEBUG]   running PYTHONPATH=/home/yangyu/OpenDroneMap/SuperBuild/install/lib/python2.7/dist-packages /home/yangyu/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all /home/yangyu/OpenDroneMap/tests/test_data/opensfm
/bin/sh: 1: /home/yangyu/OpenDroneMap/SuperBuild/src/opensfm/bin/run_all: not found
Traceback (most recent call last):
  File "/home/yangyu/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/home/yangyu/OpenDroneMap/scripts/opensfm.py", line 83, in process
    (context.pyopencv_path, context.opensfm_path, tree.opensfm))
  File "/home/yangyu/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 127

bryanibit, 2017-01-26:
I use command "bash run.sh --project-path ./test/test_data"

bryanibit, 2017-01-27:
[logconfig.txt](https://github.com/OpenDroneMap/OpenDroneMap/files/734264/logconfig.txt)

This is the result of "bash configure.sh >> logconfig.txt"

bryanibit, 2017-01-27:
Maybe no found "openni"

bryanibit, 2017-01-29:
@dakotabenjamin  Could you please tell me why it said no found openni and how to solve it?

dakotabenjamin, 2017-01-30:
openni is an optional package that doesn't need to get built, I think. Have you tried rerunning now?

bryanibit, 2017-02-13:
Sorry to reply you so later.  I found there are errors in the process of "bash configure.sh". 
//CMake Error at /home/yangyu/OpenDroneMap/SuperBuild/build/catkin/stamp/download-    //catkin.cmake:27 (message):
 //  error: downloading 'https://github.com/ros/catkin/archive/0.6.16.zip
 //  failed
 //    status_code: 1
 //    status_string: "Unsupported protocol"
 //   log: Protocol "https" not supported or disabled in libcurl
 // Closing connection -1
The same things happened in 3.2.5.tar.bz2 python-wrapper.zip. Even though I replace them with correct files and place these correct files in folder download, the errors is showed the same and the original correct files is deleted and the size of the new files with the same names with deleted files is zero.

bryanibit, 2017-02-14:
When I rerun the configure.sh, the same errors are shown. The problem is similar to #427. Some files can not be downloaded because of "https". However, I copy the right files downloaded manually to the Superbuild/download folder. This can not solve it.

dakotabenjamin, 2017-02-14:
Are you able to enable transfer over secure http?

bryanibit, 2017-02-15:
How to prove the https works ok?

dakotabenjamin, 2017-02-21:
Can you access any websites using https?

bryanibit, 2017-02-22:
Https website can be reached in Firefox browser.

dakotabenjamin, 2017-02-22:
What OS are you using? Can you tell my your CMake version? 

You may need to recompile curl with SSL support: http://stackoverflow.com/questions/29816529/unsupported-protocol-while-downlod-tar-gz-package

bryanibit, 2017-02-22:
I use Ubuntu 14.04 with ROS. Cmake is 3.6.3. I suspect maybe Chinese fire wall is the main trouble. I will try your link. Thank you.

bryanibit, 2017-03-02:
Thank you for your suggestion. It is really because curl shipped with cmake include not ssl. We need to recompile cmake with ssl. At the same time, I met the same problem with #482 about hash dismatch which can be solved through lowercase url_md5 letters.

bryanibit, 2017-03-02:
I was using VM Ubuntu 14.04 with ROS indigo

closed 2017-03-02 by bryanibit