OpenDroneMap/ODM #461: Reduce docker image size

452, 2017-01-26:
please reduce docker image size if this possible
docker image must be maximum clear, must have no includes .git .logs installation or updates caches unused libs etc
4.4 GB it's like OS))

dakotabenjamin, 2017-01-30:
@pierotofy can we build a docker images from the latest release? https://github.com/OpenDroneMap/OpenDroneMap/releases/tag/v0.2

pierotofy, 2017-01-30:
This has already been updated a few hours ago: https://hub.docker.com/r/opendronemap/opendronemap/

I think @mojodna was working on a minimal opendronemap build. Perhaps he can make some suggestions as to what can be trimmed.

mojodna, 2017-01-30:
It's here: https://github.com/mojodna/docker-opendronemap-minimal

It uses the existing `Dockerfile` to build everything, then copies the relevant bits into a new image that only contains runtime dependencies. The resulting image is ~400MB or so.

I haven't touched it in a few months, so some of the runtime dependencies will probably need to be updated. It should also be combined with @pierotofy's tweaks for CPU compatibility (so it'll run on older CPUs than the one it was built on).

pierotofy, 2017-01-30:
I wonder if there might be a way of simply adding a few intermediate steps to remove unnecessary files instead of creating a separate image.

- Removing /code/.git will reduce the size by 1.1G.
- Removing /code/download will trim another 200mb
- Removing /code/SuperBuild/src/catkin{,ceres,cmvs,ecto,gflags,mvstexturing,opencv,opengv,pcl,pdal} will free another ~500 mb
- Removing /code/SuperBuild/build/catkin{,ceres,cmvs,ecto,gflags,mvstexturing,opencv,opengv,opensfm,pcl} will free another ~500 mb

I'm going off of @mojodna's dockerfile for this list. Haven't tried to see if it breaks things.

Most of the other space is taken in /usr/lib, which is due to the many dependencies required for compilation.

I wonder if I could take the list here https://github.com/OpenDroneMap/OpenDroneMap/blob/master/packages.Dockerfile and simply remove the unneeded packages.

dakotabenjamin, 2017-02-08:
I'm starting to dive into this. Here is the size of each layer for the built image, after some edits (remove .git, squash all apt-get to one RUN command): 
<pre>
$ docker history odm_image
IMAGE               CREATED             CREATED BY                                      SIZE                COMMENT
305cb8e72378        15 minutes ago      /bin/sh -c #(nop)  ENTRYPOINT ["python" "/cod   0 B
<b>db92bd857233        15 minutes ago      /bin/sh -c cd SuperBuild && mkdir build && cd   1.718 GB</b>
b3acc2d18167        About an hour ago   /bin/sh -c #(nop) COPY dir:b3e4c51c11fd93725b   6.273 kB
67ea163d9b0f        About an hour ago   /bin/sh -c #(nop) COPY file:c1186e1921bb28e06   3.464 kB
7ef53497e475        About an hour ago   /bin/sh -c #(nop) COPY dir:744fb8cab953bc8364   17.02 kB
e2defd919549        About an hour ago   /bin/sh -c #(nop) COPY dir:9abcbf68f5379a6d96   64.78 kB
38e487976ecf        About an hour ago   /bin/sh -c #(nop) COPY file:2fac973761caf0933   1.75 kB
6eaa3c013703        About an hour ago   /bin/sh -c #(nop) COPY dir:29f87e1335be412805   453.3 kB
b70c0c230788        About an hour ago   /bin/sh -c #(nop) COPY dir:6cd5eb354afb8652f5   40.9 kB
d564c48d93f4        About an hour ago   /bin/sh -c #(nop) COPY dir:6834d3f462c2e1b430   310.1 kB
7a3d35ad0548        About an hour ago   /bin/sh -c #(nop) COPY file:70af5c08dc8726a4f   100 B
93182eb719ea        About an hour ago   /bin/sh -c #(nop) COPY file:64ad8b9dad40ea713   219 B
c63c7ae5bd0d        About an hour ago   /bin/sh -c #(nop) COPY file:21d6a2ef508a4212f   3 kB
dbdafcb203eb        About an hour ago   /bin/sh -c #(nop) COPY file:a3305fe610946e0ca   565 B
21680188bbbd        About an hour ago   /bin/sh -c #(nop) COPY file:79488886e6ac27325   679 B
8c6b59938438        About an hour ago   /bin/sh -c #(nop) WORKDIR /code                 0 B
4af29eb97e72        About an hour ago   /bin/sh -c mkdir /code                          0 B
6b6b5d5145f4        About an hour ago   /bin/sh -c #(nop)  ENV LD_LIBRARY_PATH=:/code   0 B
37bac355ec3a        About an hour ago   /bin/sh -c #(nop)  ENV PYTHONPATH=:/code/Supe   0 B
b55d075f2c44        About an hour ago   /bin/sh -c #(nop)  ENV PYTHONPATH=:/code/Supe   0 B
<b>a56bc81a69ea        About an hour ago   /bin/sh -c apt-get update     && apt-get inst   1.176 GB</b>
dddd482e90e7        About an hour ago   /bin/sh -c #(nop)  ENV DEBIAN_FRONTEND=nonint   0 B
a3312d80e663        About an hour ago   /bin/sh -c #(nop)  MAINTAINER Alex Hagiopol <   0 B
</pre>

dakotabenjamin, 2017-04-04:
We've trimmed more than 1GB as of #531 . Future improvements will include a full package audit, which will become a new issue.

closed 2017-04-04 by dakotabenjamin