OpenDroneMap/ODM #462: SuperBuild/build/pdal/bin/pdal: not found Error

aganjoo, 2017-01-28:
While processing the data for a test mission, the process exit with the following error messages.
Any suggestions

[DEBUG]   running /root/OpenDroneMap/SuperBuild/build/pdal/bin/pdal pipeline -i /root/node-OpenDroneMap/data/b0c57ae9-e64b-4347-9408-bde3f5de861f/odm_georeferencing/pipeline.xml --readers.ply.filename=/root/node-OpenDroneMap/data/b0c57ae9-e64b-4347-9408-bde3f5de861f/odm_georeferencing/odm_georeferenced_model.ply --writers.las.filename=/root/node-OpenDroneMap/data/b0c57ae9-e64b-4347-9408-bde3f5de861f/odm_georeferencing/odm_georeferenced_model.ply.las
/bin/sh: 1: /root/OpenDroneMap/SuperBuild/build/pdal/bin/pdal: not found
Traceback (most recent call last):
  File "/root/OpenDroneMap/run.py", line 55, in <module>
    plasm.execute(niter=1)
  File "/root/OpenDroneMap/scripts/odm_georeferencing.py", line 150, in process
    tree.odm_georeferencing_pdal)
  File "/root/OpenDroneMap/opendm/types.py", line 208, in convert_to_las
    '--writers.las.filename={f_out}'.format(**kwargs))
  File "/root/OpenDroneMap/opendm/system.py", line 28, in run
    raise Exception("Child returned {}".format(retcode))
Exception: Child returned 127

dakotabenjamin, 2017-01-30:
Have you tried rerunning ./configure.sh? Maybe an error occurred during install.

pierotofy, 2017-01-30:
See https://github.com/OpenDroneMap/WebODM/issues/85, looks like the same problem. Check your GDAL version.

pierotofy, 2017-01-30:
Oh, didn't see who posted... this can probably be closed since we've solved it.

dakotabenjamin, 2017-01-30:
Great!

closed 2017-01-30 by dakotabenjamin