OpenDroneMap/ODM #463: Eyefish images

frasanz, 2017-01-29:
I'm trying to use opendronamap with this bebop 2 imageset (180º eyefish images)

https://support.pix4d.com/hc/en-us/articles/209620783-Example-Datasets-Available-for-Download-Parrot-Bebop-2-House#gsc.tab=0

Using by default run.py without extra parameters, I have the typicall white walls, but also a very ugly model

https://skfb.ly/ZJWv

[image]

Not sure if it's a problem with the eyefish images or only I need to insert some extra parameters.

frasanz, 2017-01-29:
Sorry...also, not sure what to insert into force-ccd

dakotabenjamin, 2017-01-30:
These will be difficult if not impossible to reconstruct currently because of the fisheye and the images have horizon/sky in them. I don't know what it would take to get this to work like pix4d.

frasanz, 2017-01-30:
Ok, thanks for your answer.

hblanken, 2017-05-18:
Could we add this into an improvement suggestion/seeking help - onto the release roadmap? There are quite a few fisheye cameras which would be a good addition for ODM. Parrot fisheye + GPS exif images would make a great input addition for ODM.

KommandorKeen, 2017-05-29:
My understanding of Bebop is the stabilisation is digital and therefore the distortion is dynamic not static.
But if you are capturing the whole image then use openCV to undistort the images (chessboard procedure) and then crop the image to more like a 90 degree view.  THEN guess suitable CCD and focal length values and input these into ODM using --forc-focal and --force-ccd.

gmaclennan, 2017-06-13:
It would be great if ODM could support images from the Parrot Bebop 2, it is a great value drone that can collect good orthophotos with the Pix4D capture app.

pierotofy, 2020-02-03:
Fish-eye cameras have been fully supported for a while now. Just use the `--camera-lens fisheye` option! :)

closed 2020-02-03 by pierotofy