OpenDroneMap/ODM #465: python run.py | ImportError: No module named 'context'

jecogeo, 2017-02-03:
Hi everyone. 
I ran the configure.sh and then tried to run `python run.py --project-path "my/path" -i "my/path/images/"`. However, I get the following error: 

Traceback (most recent call last):
  File "run.py", line 4, in <module>
    from opendm import config
  File "/etc/OpenDroneMap-v0_2/opendm/config.py", line 2, in <module>
    import context
ImportError: No module named 'context'

I'm not sure, but I'm wondering if there is something wrong with the installation. Looking at the several lines of output during configure.sh execution, I noticed many many lines like this one:

-- The imported target "vtkParseJava" references the file
   "/usr/bin/vtkParseJava"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/vtk-5.10/VTKTargets.cmake"
but not all the files it references.

Ideas?

PS.: Some background: Ubuntu 16.04 LTS, Python 3.5.2 |Anaconda custom (64-bit)

dakotabenjamin, 2017-02-06:
ODM relies on Python 2.7 still. Maybe that is the problem?

dakotabenjamin, 2017-04-03:
Yes. there are issues with using ODM and Anaconda. Please reopen if you have any more questions.

closed 2017-04-03 by dakotabenjamin