OpenDroneMap/ODM #466: ODM OrthoPhoto that is produced is of very low quality relative to dataset

piousgreek, 2017-02-04:
I am processing a small survey of 40 geotagged pictures and the output orthomosaic is of very low resolution and I would like to correct this.  The survey was taken from a GoPro Hero 4 Black that has been modified with a rectilinear 3.97 mm lens.   I'm running ODM version 2 58 from the command line on Ubuntu 14.04 LTS with standard settings as follows: "python run.py --Project-Path /path/to/project".  

The meshes that it produces are very good, and comparable to the ODM examples, however the OrthoMosaic is of very low resolution although it is stitched together quite well.

The ODM orthophoto can be found here:
https://www.dropbox.com/s/4famf2xlxgs6wxb/odm_orthophoto.tif?dl=0

The geotagged source images can be found here:
https://www.dropbox.com/s/f1vvqos4l6cfxny/ShepherdDeWolf.zip?dl=0

Let me know if you need any logs.

pierotofy, 2017-02-04:
Try to tweak the resize-to and orthophoto-resolution parameters.

`python run.py --help`

smathermather, 2017-02-05:
Orthophoto parameters matter more -- resize-to shouldn't effect the final orthophoto resolution, just the matching process because texture comes from full resolution images.

piousgreek, 2017-02-06:
Thank you both... Looks like I also need to force-focal since the gopro exif data doesn't reflect the rectilinear lens mod.

If you want to add the geotagged files I uploaded to the example database you're welcome to.

I tweaked some other parameters, we'll see how long it takes and how it goes:

python run.py --orthophoto-resolution 1.0 --force-focal 3.97  --pmvs-csize 1 --mesh-size 400000 --mesh-octree-depth 12 --project-path /Desktop/quickwork/geotagged

Thanks again!

piousgreek, 2017-02-06:
I reran the data set with the "--orthophoto-resolution 1.0" parameter and the odm_orthophoto created were of even lesser resolution!

[image]

The complete command I ran was as follows: 

`python run.py --orthophoto-resolution 1.0 --force-focal 3.97 --pmvs-csize 1 --mesh-size 400000 --mesh-octree-depth 12 --project-path /Desktop/quickwork/geotagged`

Some of the warnings I see are:

[WARNING] Initial residual too low: 4.1068e-07 < 0.000001
[WARNING] Initial residual too low: 1.55759e-07 < 0.000001
[WARNING] Initial residual too low: 6.97806e-07 < 0.000001
Warning: Zero-length normals detected: 342 face normals, 1 vertex normals
[WARNING] No coordinates file. Generating coordinates file: /home/siddhartha/Desktop/quickwork/geotagged/odm_georeferencing/coords.txt
Warning 6: Driver GTiff does not support NUM_THREADS creation option

dakotabenjamin, 2017-02-06:
Orthophoto resolution is calculated in pixels/meter, so decreasing it from 20 (default) to 1 would decrease the resolution. 

For the other warnings:
first three are fine, ignore them,
the zero-length normals is for texturing? I'm not sure about that one but I've never actually had a problem with it before
the coordinates one shouldn't really even be a warning, but rather an INFO message
The last one can be removed by upgrading to GDAL 2

pierotofy, 2017-02-06:
I think you need to increase the --orthophoto-resolution parameter, not decrease it.

"Orthophoto Ground resolution in pixels / meter". You want to increase the number of pixels per meter.

dakotabenjamin, 2017-02-06:
thanks @pierotofy !

pierotofy, 2017-02-06:
Ah, you beat me to it @dakotabenjamin :laughing:

piousgreek, 2017-02-06:
You're right, i totally misinterpreted that parameter... i was thinking centimeters per pixel despite the fact that the help files says that the parameter expected is given in meters per pixel...  Thanks again guys

closed 2017-02-06 by piousgreek

moscheja, 2018-05-12:
Why is quality of the result so bad. Compared to the result of Dronedeploy.com the result is so much bader, that it is completey useless:

- GeotIFF has a lot of failures, distortions, is not cleat etc.
- 3D model is useless
- measures are completey useless, failures are about 2 m wrong ????

I know it's free and DroneDeploy not but what shall I do with a software which doesn't work....

yjmenezes, 2018-05-12:
@moscheja 

Do you mind to send me the DJI Mavic flight log ?
I do not know DJI hardware but i believe it outputs some information from IMU.

I apologise if i am rude.
Free do not mean "no charge", it means "freedom".
People work hard on ODM, i believe as a volunteer work, i do not know if ODM has some sponsor. Current version is 0.3.1. The good of FOSS is that anyone can see the source code, learn, and try to help to make it better.

cheers

coopermor, 2019-06-08:
For people finding this issue the unit of ----orthophoto-resolution has now changed to cm / pixel as of #889 

The default value is now 5.0 and decreasing it increased the quality.

rdwrt, 2019-07-29:
Hi, we just discovered that the manual at https://docs.opendronemap.org/using.html#creating-high-quality-orthophotos now seems inconsistent. The unit seems to be changed, but increasing this unit now would likely lower the resolution whereas the manual still seems to say that it increases the resolution: 

> --orthophoto-resolution is the resolution of the orthophoto in cm/pixel. Increase this value for a higher resolution result.

pierotofy, 2019-07-29:
Nice catch, yes that's a mistake. It should be that lowering the value increases the resolution. Could you fix it by editing https://github.com/OpenDroneMap/docs/blob/publish/source/using.rst ?