OpenDroneMap/ODM #467: Under-utilization of objects

dakotabenjamin, 2017-02-07:
In the old days of ODM it was basically an IO toolchain to loosely connected C++ modules. We've greatly improved the code, adding objects like ODM_Photo, ODM_Reconstruction, ODM_GCPoint, ODM_Georef, and ODM_Tree. Even still, ODM_Reconstruction initializes a base class that doesn't even exist. Even though the ODM_Photo object exists, modules simply refer to and load the images themselves instead of calling the object. 

I'd like to open up discussion on how to improve and better integrate the object code into the whole workflow process so we can best utilize the code.

GeospatialDaryl, 2017-02-07:
Hi-

I looked at this when I refactored an original version of run.py.  I was
hoping to instigate a conversation about this at that time, rather than
tacking on new features to that design.  It looks like some of the OO
suggestions I made have persisted.

From my perspective, we need an object model for reconstructions, images,
point sets, etc, that support dynamic interaction with a 'solution'.  I'd
be happy to contribute to that discussion, but it takes a fair amount of
coordination for the community to agree to an object model, then move
forward on the implementation of that in code.

So - I whole-heatedly agree.  This would be a great time to define an
object model, and refactor the application to support that.  I think it
would facilitate the Web interface as well.

Cheers-

Daryl

dakotabenjamin, 2017-02-08:
I believe when @edgarriba began work on the OO side of his work, he used your suggestions as a basis. Can you explain further what you mean by dynamic interaction with a 'solution'? With the development of an API, greater interaction from the user becomes possible. A good object model would facilitate that.

edgarriba, 2017-02-08:
hi! yes ideally for GUIs or whatever interaction a good OO design is needed and that's what I intended to do. You could try to mimic some base classes or at least just simplify from it OpenSfm
https://github.com/mapillary/OpenSfM/blob/master/opensfm/types.py

dakotabenjamin, 2017-02-08:
That's a really good idea. We could start by inheriting some of those classes into the ones already built.

smathermather, 2020-04-14:
Lots of refactoring went into play in Spring of 2019. I will close this.

closed 2020-04-14 by smathermather