OpenDroneMap/ODM #468: Textured Model up vector is not consistent

pierotofy, 2017-02-07:
When processing the following datasets, the orientation of the textured model is not consistent and at times, is flipped upside down.

Processing this one for example, seems to align correctly: https://github.com/pierotofy/drone-datasets/tree/master/sheffield_park2

While processing this one, the model is oriented upside-down: https://github.com/pierotofy/drone-datasets/tree/master/sheffield_park

rionlerm, 2017-02-08:
Had similar issues with flipping before. Not always a problem though.

smathermather, 2017-02-08:
Reviewing further, this seems to be an issue with WebODM using the non-geo version of the model (which will be arbitrarily oriented).

pierotofy, 2017-02-08:
Confirmed that this is an issue with WebODM loading the non-geo version of the model.

This has been fixed in the unittest branch which will be merged into master by end of week.

closed 2017-02-08 by pierotofy

smathermather, 2017-02-08:
Wow! Awesome.